project-chip/connectedhomeip #30507: [1.2] I want to make chip-tool as an OTA provider. But I am facing error related to ACL.

Shahmeera, 2023-11-16:
### Reproduction steps

For generating chip-tool as a OTA provider I have enabled OTA-provider cluster as a server from ZAP tool. So the process would be like below
(1) Commission OTA requestor from chip-tool
(2) Announce OTA provider(chip-tool) for presence. For this step I knew chip-tool node ID would be 112233.
Now I am facing only one error here in the function ProcessCommandDataIB (/src/app/CommandHandler.cpp). Here I am getting error log like this **acl err code is src/access/AccessControl.cpp:491: CHIP Error 0x000000A5: The CHIP message is not granted access**

`VerifyOrExit(mExchangeCtx && mExchangeCtx->HasSessionHandle(), err = CHIP_ERROR_INCORRECT_STATE);

    {   
        ChipLogDetail(DataManagement,"cluster %x and endpoint %d\n",concretePath.mClusterId,concretePath.mEndpointId);
        Access::SubjectDescriptor subjectDescriptor = GetSubjectDescriptor();
        Access::RequestPath requestPath{ .cluster = concretePath.mClusterId, .endpoint = concretePath.mEndpointId };
        Access::Privilege requestPrivilege = RequiredPrivilege::ForInvokeCommand(concretePath);
        err                                = Access::GetAccessControl().Check(subjectDescriptor, requestPath, requestPrivilege);
        ChipLogError(DataManagement, "acl err code is %" CHIP_ERROR_FORMAT, err.Format());
        if (err != CHIP_NO_ERROR)
        {
            if (err != CHIP_ERROR_ACCESS_DENIED)
            {
                return FallibleAddStatus(concretePath, Status::Failure) != CHIP_NO_ERROR ? Status::Failure : Status::Success;
            }
             ChipLogDetail(DataManagement,"TODO wildcard\n");
            // TODO: when wildcard invokes are supported, handle them to discard rather than fail with status
            return FallibleAddStatus(concretePath, Status::UnsupportedAccess) != CHIP_NO_ERROR ? Status::Failure : Status::Success;
        }  
    }`

``
    The reason behind this error is I can not assign access control to the chip-tool(ota provider).
    
    ./chip-tool accesscontrol write acl '[{"fabricIndex": 1, "privilege": 5, "authMode": 2, "subjects": [112233], "targets": null}, {"fabricIndex": 1, "privilege": 3, "authMode": 2, "subjects": null, "targets": [{"cluster": 41, "endpoint": null, "deviceType": null}]}]' 112233 0
I want to assign privileges to the chip-tool in my case(chip-tool+ota-provider) **112233** 
Please help me to manipulate the code to add another ACL entry of privilege -3 authmode-2 and subject-null.

### Bug prevalence

none

### GitHub hash of the SDK that was being used

https://github.com/project-chip/connectedhomeip/tree/v1.2.0.1

### Platform

IMX8

### Type

Test Improvement

bzbarsky-apple, 2023-11-17:
chip-tool does not currently include an ACL provider.  You would need to make changes to it to do that....

closed 2023-11-17 by bzbarsky-apple

Shahmeera, 2023-11-17:
If I comment ACL check function I am able to reach to the query image request and response, but packet transfer is not happening. And according to my understanding it is only due to ACL entries on chip-tool(112233 which is OTA provider in my case). Am I right or wrong?
If ACL is must then I want to add entries  `CreateEntry(const SubjectDescriptor * subjectDescriptor, FabricIndex fabric, size_t *index,const Entry & entry` in  function or before this function.

`CHIP_ERROR AccessControl::Check(const SubjectDescriptor & subjectDescriptor, const RequestPath & requestPath,
                                Privilege requestPrivilege)`

bzbarsky-apple, 2023-11-18:
> but packet transfer is not happening

Well, is chip-tool actually implementing BDX bits?

> If ACL is must then I want to add entries

It depends on what ACL provider you use....  You can directly implement the ACL check; see `src/app/dynamic_server/AccessControl.cpp` for an example.

Shahmeera, 2023-11-20:
yes, If I comment this lines in the function `Status CommandHandler::ProcessCommandDataIB(CommandDataIB::Parser & aCommandElement)` I would be able to reach up to the Query image request and response.

[1700455690.256932][1900900:1900902] CHIP:DMG: **AccessControl: denied**
[1700455690.256953][1900900:1900902] CHIP:DMG: **acl err code is src/access/AccessControl.cpp:491: CHIP Error 0x000000A5: The CHIP message is not granted access**
[1700455690.256974][1900900:1900902] CHIP:DMG: Received command for Endpoint=0 Cluster=0x0000_0029 Command=0x0000_0000
[1700455690.257053][1900900:1900902] CHIP:ZCL: OTA Provider received QueryImage
[1700455690.257066][1900900:1900902] CHIP:ZCL:   VendorID: 0xfff1
[1700455690.257080][1900900:1900902] CHIP:ZCL:   ProductID: 32769
[1700455690.257094][1900900:1900902] CHIP:ZCL:   SoftwareVersion: 1
[1700455690.257107][1900900:1900902] CHIP:ZCL:   ProtocolsSupported: [
[1700455690.257123][1900900:1900902] CHIP:ZCL:     0
[1700455690.257137][1900900:1900902] CHIP:ZCL:   ]
[1700455690.257151][1900900:1900902] CHIP:ZCL:   HardwareVersion: 0
[1700455690.257165][1900900:1900902] CHIP:ZCL:   Location: XX
[1700455690.257179][1900900:1900902] CHIP:ZCL:   RequestorCanConsent: 0
[1700455690.257196][1900900:1900902] CHIP:SWU: parse ota header:

I have seen this ACL check `src/app/dynamic_server/AccessControl.cpp` but fot that also I need to add entries like 
`"privilege": 3, "authMode": 2, "subjects": null,` etc.

Entry newentry;
FabricIndex fabric = 1;
newentry.SetAuthMode(AuthMode::kCase);
size_t index;  */
CHIP_ERROR result = CreateEntry(&subjectDescriptor,  fabric, &index, newentry);

 I have tried to manually add ACL entry but ` SuccessOrExit(err = entry.GetAuthMode(authMode));` this is giving an error.