project-chip/connectedhomeip #30508: [Feature] Water Content Measurement

goronlant, 2023-11-16:
### Feature description

Due to my personal project, I want to develop a soil moisture device for my plants. In the document "Matter 1.1 Application Cluster Specification" under 2.6. Water Content Measurement, it states that there should be a cluster called Soil Moisture Measurement with the identifier 0x0408. The issue is that it is not found anywhere in the code. Does the feature exist, or is it still to come?

If there are alternative ways to implement this, I would greatly appreciate it if you could provide that information.

Thanks in advance.

### Platform

esp32

### Platform Version(s)

1.1

### Anything else?

I am uploading a screenshot from the document.

[image]

bzbarsky-apple, 2023-11-17:
It looks like https://github.com/CHIP-Specifications/connectedhomeip-spec/pull/7208 removed the provisional marking, but there is indeed no implementation, so I have no idea how that was supposed to work.  @ReneJosefsen

ReneJosefsen, 2023-11-17:
Yeah, this one is in the same bucket as with the other alias clusters but it looks like they slipped under the radar 😔 The idea is/was that alias clusters could use the same test plans hence not being provisional once one of the clusters was certified, but it really never ended up this way. For Concentration Measurement and Resource Monitoring we ended up removing these: 

https://github.com/CHIP-Specifications/connectedhomeip-spec/pull/7289
https://github.com/CHIP-Specifications/connectedhomeip-spec/pull/7290

So this simply needs to happen for these as well, given there is no SDK implementation or test plan for these clusters. Sorry for my mistake here.

PSONALl, 2023-11-17:
@ReneJosefsen are there any plans that these clusters will be implemented in the sdk?

ReneJosefsen, 2023-11-17:
I'm not aware of any. In the current state, unless anybody is of another opinion, based on how this was handled in the other alias clusters, for now these simply needs to be removed from the spec. If anybody wants to have it added back in and get implemented, I guess it is a matter of following the process for doing so and we add these back into the spec and create test plans and so on.

ReneJosefsen, 2023-11-17:
Another issue if anybody wants to bring this to the market, is that I do not see any device types actually using the Leaf and Soil clusters. So technically speaking there is really no device to certify at the moment.

goronlant, 2023-11-17:
Thank you for your response. 
Bad news for me. What's the reason for removing this feature? The smart garden represents a significant market and opportunity for Matter.

ReneJosefsen, 2023-11-17:
Well, for now since it is not implemented in the SDK and is not certifiable, it should either not be in the spec or be marked as provisional. The agreement done with other alias clusters is to remove them, also given that no device types are using these clusters. If companies want to bring devices like this to the market, (simplified) they can follow the process in CSA to get it approved and then the clusters can be added back in again together with any suitable device types.

dnicoara, 2024-02-13:
I've got a similar personal project. I'm using relative humidity for now since it's still in the spec.

Not as ideal, but relative humidity is supported by ecosystems from an integration point of view so I can monitor it without creating my own fabric/app/etc.