project-chip/connectedhomeip #30512: Unauthorised access to Matter accessories warning

Tanjchy3, 2023-11-16:
### Reproduction steps

Enter manual code
Click on the button Go
Observe dialog

[image]

Question:
What does this warning mean? How to remove the warning?

### Bug prevalence

Every time

### GitHub hash of the SDK that was being used

/

### Platform

esp32

bzbarsky-apple, 2023-11-17:
@Tanjchy3 Is this on iOS or Android or somewhere else?  I doubt that UI is running on esp32, as you claim in "Platform".

woody-apple, 2025-04-02:
This is a result of installing the apple developer profile and is expected, you can learn more about the profile here:
https://github.com/project-chip/connectedhomeip/blob/master/docs/guides/darwin.md#profile-installation

closed 2025-04-02 by woody-apple