project-chip/connectedhomeip #30527: [ICD] missing dependenc/wrong dependency from ReliableMessageMgr.cpp

andy31415, 2023-11-16:
In https://github.com/project-chip/connectedhomeip/blob/master/src/messaging/ReliableMessageMgr.cpp#L41 we have:

```
#if CHIP_CONFIG_ENABLE_ICD_SERVER
#include <app/icd/ICDManager.h>  // nogncheck
#include <app/icd/ICDNotifier.h> // nogncheck
#endif
```

which includes ICDManager, however dependencies in https://github.com/project-chip/connectedhomeip/blob/master/src/messaging/BUILD.gn#L70 only include `src/app/icd:notifier`

This is not ok, if you include ICDManager you must also depend on `:manager`. Using `//nogncheck` is a bad "workaround" (and it was used previously just because ifdefs ... it does not apply here).

Logic has to be changed to:
  - depend on what you use
  - dependency to ICD should be done based on CHIP_CONFIG_ENABLE_ICD_SERVER not always as it is currently

jmartinez-silabs, 2023-11-16:
`//nogncheck` was not used as a bad workaround, ICDManager was conditionnally included in the dependency of the messaging static_lib

```
  if (chip_enable_icd_server) {
    public_deps += [ "${chip_root}/src/app/icd:manager" ]
  }
```

It should still be as it is used at line 262
`mrpBackoffTime += app::ICDManager::GetFastPollingInterval();`

it seems it was removed by mistake by someone.

So the solution is to add back the dependency in the build.gn. Or remove the include and change how the `FastPollingInterval` value is retrieved for the `mrpBackoffTime` calculation

andy31415, 2023-11-16:
Should icd:notifier have also been part of CHIP_ENABLE_ICD_SERVER?

andy31415, 2023-11-16:
The reason was probably:

```
2023-11-16 20:59:25 INFO    ERROR Dependency cycle:
2023-11-16 20:59:25 INFO      //third_party/connectedhomeip/src/app/icd:manager ->
2023-11-16 20:59:25 INFO      //third_party/connectedhomeip/src/app/icd:sender ->
2023-11-16 20:59:25 INFO      //third_party/connectedhomeip/src/lib/address_resolve:address_resolve ->
2023-11-16 20:59:25 INFO      //third_party/connectedhomeip/src/messaging:messaging ->
2023-11-16 20:59:25 INFO      //third_party/connectedhomeip/src/app/icd:manager
```

Address_resolve depends on messaging which depends on ICD.
Our layering is wrong.

andy31415, 2023-11-16:
It looks to me like #30385 did this.

jmartinez-silabs, 2023-11-16:
Yeah this circular dependency is newly added code it seems. I can clean this up by remove icd:manager from :messaging and creating a new icd:dataclass 

For the `icd:notifier.` It is a clean dependency, I can gate it in #ifdef CHIP_ENABLE_ICD_SERVER if you prefer. It is mostly a virtual call with a class to register and broadcast to listeners.

andy31415, 2023-11-16:
The reason for AddressResolve dependency seems to be `#include <messaging/ReliableMessageProtocolConfig.h>`.

We could maybe split that out in a separate source set.

jmartinez-silabs, 2023-11-16:
Also and option. I have some time tomorrow to clean that up

jmartinez-silabs, 2023-11-17:
This PR https://github.com/project-chip/connectedhomeip/pull/30552/files#diff-e876b9bac9dbe61d8a74bf89603d90d47850fef5167ca794d37fd58f17ea3c3d
remove the ICDManager Include and its use in favour of the definition for the time being.

I will now work on the AddressResolve dependency we discussed, And @mkardous-silabs should have already started working on the icd Data Class I mentioned.

PR #30618 (2023-11-28): [ICD] Create ICDConfigurationData class with associated refactor
#### Description
Implementation of the ICD feature set created a circular dependencie issue which was caused by the ICDManager needing to send Check-In messages and being the owner of the Fast Poll configuration that is used by the ReliableMessageMgr.
This was also a problem for the ReliableMessageProtocolConfig which requires the Fast poll config, slow poll config owned by the ICD Manager"

To circumvent the issue, this PR adds the ICDData class that will own all the configurations. This will allow all the consumers to depend on [... cut]
closed 2023-11-28 by mkardous-silabs