project-chip/connectedhomeip #30542: [BUG] Android project error for ble connect

freemanspace, 2023-11-17:
### Reproduction steps

Android project

do commission flow

deviceController = ChipClient.getDeviceController(context)
deviceController.pairDevice(gatt, connId, deviceId, mDeviceInfo!!.setupPinCode, network)

first time pair will receive error code 50 (wifi setting issue) with GenericChipDeviceListener.onCommissioningComplete(nodeId: Long, errorCode: Int) 

than i reset device and do commission flow again

deviceController.pairDevice(gatt, connId, deviceId, mDeviceInfo!!.setupPinCode, network)

i will receive error "Bluetooh connection already in use" with GenericChipDeviceListener.onError(error: Throwable?)

why??

### Bug prevalence

every time

### GitHub hash of the SDK that was being used

73c327e91a4252daac96728452e8cb713ea83697

### Platform

android

freemanspace, 2023-11-20:
i try do 
before pairDevice fun
 do 
deviceController.close()
than
deviceController = deviceController = ChipClient.getDeviceController(context)
deviceController.pairDevice(gatt, connId, deviceId, mDeviceInfo!!.setupPinCode, network)
will crash
log===>
2023-11-20 11:31:58.403 20018-20018 AndroidRuntime          com.test                     E  FATAL EXCEPTION: main
                                                                                                    Process: com.test, PID: 20018
                                                                                                    chip.devicecontroller.ChipDeviceControllerException: src/transport/raw/BLE.cpp:118: CHIP Error 0x0000000B: No memory
                                                                                                    	at chip.devicecontroller.ChipDeviceController.pairDevice(Native Method)
                                                                                                    	at chip.devicecontroller.ChipDeviceController.pairDevice(ChipDeviceController.java:160)
                                                                                                    	at chip.devicecontroller.ChipDeviceController.pairDevice(ChipDeviceController.java:129)
                                                                                                    	at com.google.chip.chiptool.SharedMatterViewModel$pairDeviceWithBle$1.invokeSuspend(SharedMatterViewModel.kt:193)
                                                                                                    	at kotlin.coroutines.jvm.internal.BaseContinuationImpl.resumeWith(ContinuationImpl.kt:33)
                                                                                                    	at kotlinx.coroutines.DispatchedTask.run(DispatchedTask.kt:106)
                                                                                                    	at android.os.Handler.handleCallback(Handler.java:883)
                                                                                                    	at android.os.Handler.dispatchMessage(Handler.java:100)
                                                                                                    	at android.os.Looper.loop(Looper.java:214)
                                                                                                    	at android.app.ActivityThread.main(ActivityThread.java:7397)
                                                                                                    	at java.lang.reflect.Method.invoke(Native Method)
                                                                                                    	at com.android.internal.os.RuntimeInit$MethodAndArgsCaller.run(RuntimeInit.java:492)
                                                                                                    	at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:952)

Trishia, 2024-01-12:
I have faced same issue about "No memory" error on second time,  "Bluetooth connection already in use" error on third time once commissioning failed at the step "FindOperational" over BLE, even if I released gatt resources such as gatt.disconnect(), gatt.close(), chipDeviceController.close() etc in onError() callback.

Allen-Guof, 2024-02-01:
Save issue.

VinhBH1, 2024-06-12:
any update , the same here

bhavin-android, 2025-03-24:
I'm also getting "**Bluetooth connection already in use**".
@freemanspace have you found a solution?