project-chip/connectedhomeip #30553: [1.3][ICD] Add Helper to find subscription with a CAT tag OR a NodeId

jepenven-silabs, 2023-11-17:
ICD Check-In Protocol allows Monitored subject to be a CaseAuthenticatedTag in order to retrieve a subscription. However it seems like we do not store any CAT tag within the read handler. Making it impossible to find a subscription from a CAT tag.

bzbarsky-apple, 2023-11-17:
Doesn't the read handler have a session?  The session knows about the CATs of the other side, no?  @jepenven-silabs

bzbarsky-apple, 2023-11-17:
In particular, see https://github.com/project-chip/connectedhomeip/blob/dca427168edd52b958c6d21cd1290e516b32a8cf/src/app/ReadHandler.h#L401 and https://github.com/project-chip/connectedhomeip/blob/dca427168edd52b958c6d21cd1290e516b32a8cf/src/access/SubjectDescriptor.h#L43-L44

mkardous-silabs, 2023-11-29:
This was resolved with #30526

closed 2023-11-29 by mkardous-silabs