project-chip/connectedhomeip #30576: [BUG] CommandHandler tries to send response to a request when not required

simonhmorris1, 2023-11-20:
### Reproduction steps

As part https://github.com/project-chip/connectedhomeip/pull/30201 a ConnectNetwork request is received but no ConnectNetworkResponse is required.  This will log errors in CommandHandler. 

Update  CommandHandler to notify it that no response is required to prevent the error being reported. See discussion here https://github.com/project-chip/connectedhomeip/pull/30201#discussion_r1398119342

`[1699270892.045538][1147:1147] CHIP:NP: Instance::OnResult Non-Concurrent Mode, ConnectNetworkResponse will NOT be sent
[1699270892.045602][1147:1147] CHIP:DMG: Endpoint 0, Cluster 0x0000_0030 update version to 869e5227
[1699270892.045639][1147:1147] CHIP:DMG: Decreasing reference count for CommandHandler, remaining 0
[1699270892.045753][1147:1147] CHIP:DMG: Failed to send command response: src/app/CommandHandler.cpp:221: CHIP Error 0x00000003: Incorrect state
[1699270892.045772][1147:1147] CHIP:DMG: Command handler moving to [AwaitingDe]`

To reproduce, run the system in Non-concurrent mode as described in the above ticket reference.

### Bug prevalence

Every time on commissionee when running in Non-concurrent mode

### GitHub hash of the SDK that was being used

cc6f964

### Platform

raspi

tehampson, 2023-11-28:
Do you mind pointing out where in the spec it mentions that ConnectNetworkResponse is not required and in fact shouldn't be responded to in this instance. I am not familuar with the Non-concurrent mode stuff so any pointers to ramp up on this would be appreciated

simonhmorris1, 2023-11-28:
Here are some pointers to the Matter 1.2 specification.

Section 5.5 indicates two modes, concurrent connection and non-concurrent connection commissioning flow. A Commissioner and Commissionee with non-concurrent connection capability cannot be simultaneously connected to both the operational network (e.g Wi-Fi) and the commissioning channel (BLE).

11.8.7.9. ConnectNetwork Command
When non-concurrent commissioning is being used by a Commissioner or Administrator, it is possible
that the only method to determine success of the operation is operational discovery of the Node
on the new operational network. Therefore, before invoking the ConnectNetwork command, the
client SHOULD re-invoke the Arm Fail-Safe command with a duration that meets the following:
3. Sufficient time to allow operational discovery on the new network by a Commissioner or
Administrator.