project-chip/connectedhomeip #30594: [Feature] Prebuilt Android CHIPTool?

olalonde, 2023-11-21:
### Feature description

I have been [trying to build](https://devzone.nordicsemi.com/f/nordic-q-a/105907/cannot-build-chiptool-for-android-clang-error-unknown-argument--fmacro-prefix-map) the Android CHIPTool for a few hours on my Macbook Air but keep getting issues during the build. I was wondering if there are prebuilt binaries somewhere I can just install on my phone? I don't need to contribute to the project, I just want to use the app on my phone... Thanks.

### Platform

android

yunhanw-google, 2023-12-09:
Hi @olalonde, we are still in discussion regarding how and where to release prebuild android binary.
for compilation, please use ubuntu or debian to compile android chip-tool with the below instruction.

git clone https://github.com/project-chip/connectedhomeip.git
cd connectedhomeip
git pull
git submodule init
git submodule update
docker run -it -v ~/workspace/connectedhomeip:/connectedhomeip [ghcr.io/project-chip/chip-build-android:28](http://ghcr.io/project-chip/chip-build-android:28)
git config --global --add safe.directory "*"
source scripts/bootstrap.sh
./scripts/build/build_examples.py --target android-arm64-chip-tool build

yunhanw-google, 2024-01-14:
dup with https://github.com/project-chip/connectedhomeip/issues/29259

closed 2024-01-14 by yunhanw-google