project-chip/connectedhomeip #30599: [NXP] Layering inversion in ConfigurationmanagerImpl.cpp

andy31415, 2023-11-21:
We have:

```
src/platform/nxp/common/ConfigurationManagerImpl.cpp
30:#include <app/server/Server.h>
```

It is not ok for something inside Platform to depend On app - this inverts layering as platform is supposed to be the lowest level and app is one of the highest levels. Generally GN would prevent this however app is excluded from layering checks because of legacy code. We should fix these layering inversions.

Found while checking #30596

andy31415, 2023-11-21:
It looks like the reason this is done is for `Server::GetInstance().GetFabricTable().DeleteAllFabrics();`.
We should figure out some eventing system and pass this into app rather than having platform calling directly into server.

marius-alex-tache, 2023-11-22:
@dinabenamar

PR #30891 (2023-12-11): [NXP] Remove circular dependency on application code in platform file ConfigurationManagerImpl.cpp
The aim of this PR is to fix #30599 layering inversion issue in 'src/platform/nxp/common/ConfigurationManagerImpl.cpp' , by removing the dependency on application code in the platform file.
closed 2023-12-11 by mergify[bot]