project-chip/connectedhomeip #30624: [BUG] [Android] crash happens when device reboot

MatteoSomensi, 2023-11-23:
### Reproduction steps

When device reboot and the application tries to send a message to the ChipClient after a number of failed attempts a native exception is thrown which causes the application to crash.

### Bug prevalence

systematic

### GitHub hash of the SDK that was being used

181b0cb14ff007ec912f2ba6627e05dfb066c008

### Platform

android

### Anything else?

Failed to Send CHIP MessageCounter:172130281 on exchange 33460i sendCount: 4 max retries: 4
Time out! failed to receive report data from Exchange: 33460i
A  java_vm_ext.cc:591] JNI DETECTED ERROR IN APPLICATION: JNI GetObjectClass called with pending exception java.lang.NullPointerException: 
                                                                                                    java_vm_ext.cc:591] (Throwable with no stack trace)
                                                                                                    java_vm_ext.cc:591] 
                                                                                                    java_vm_ext.cc:591]     in call to GetObjectClass
A  runtime.cc:691] Runtime aborting...

yunhanw-google, 2023-11-28:
is it crash happening in android controller?

MatteoSomensi, 2023-11-30:
Appens when call this one in my ChipClient
/**
     * Wrapper around [ChipDeviceController.getConnectedDevicePointer] to return the value directly.
     */
    suspend fun getConnectedDevicePointer(nodeId: Long): Long {
        return suspendCancellableCoroutine { continuation ->
            getDeviceController()
                .getConnectedDevicePointer(
                    nodeId,
                    object : GetConnectedDeviceCallback {
                        override fun onDeviceConnected(devicePointer: Long) {
                            continuation.resume(devicePointer)
                        }

                        override fun onConnectionFailure(nodeId: Long, error: Exception) {
                            val errorMessage = "Unable to get connected device with nodeId $nodeId"
                            continuation.resumeWithException(IllegalStateException(errorMessage))
                        }
                    }
                )
        }
    }

pgatti86, 2023-12-22:
Following @MatteoSomensi issue, when a matter device falls out of network reachability the chip library try to resend the command. For example if I try to obtain a device pointer or read/write a cluster attribute. The library execute some retries and then throws and exception. Such exception is not catchable in any way and the app crashes.

Here a detailed log of an attempt to read this attribute: readTemperatureSetpointHoldDurationAttribute

**Is there any way to catch those crash in android app?**

09:07:26.127  E  Failed to Send CHIP MessageCounter:180467474 on exchange 63959i sendCount: 4 max retries: 4
09:07:29.189  E  Time out! failed to receive report data from Exchange: 63959i
09:07:29.212  E  readTemperatureSetpointHoldDurationAttribute failure
                 chip.devicecontroller.ChipClusterException: CHIP cluster error: 1
09:07:31.403  A  java_vm_ext.cc:596] JNI DETECTED ERROR IN APPLICATION: JNI NewStringUTF called with pending exception java.lang.NullPointerException: 
                 java_vm_ext.cc:596] (Throwable with no stack trace)
                 java_vm_ext.cc:596] 
                 java_vm_ext.cc:596]     in call to NewStringUTF
09:07:31.579  E  kotlinx.coroutines.TimeoutCancellationException: Timed out waiting for 5000 ms (Ask Studio Bot)
                 	at kotlinx.coroutines.TimeoutKt.TimeoutCancellationException(Timeout.kt:191)
                 	at kotlinx.coroutines.TimeoutCoroutine.run(Timeout.kt:159)
                 	at kotlinx.coroutines.EventLoopImplBase$DelayedRunnableTask.run(EventLoop.common.kt:501)
                 	at kotlinx.coroutines.EventLoopImplBase.processNextEvent(EventLoop.common.kt:280)
                 	at kotlinx.coroutines.DefaultExecutor.run(DefaultExecutor.kt:109)
                 	at java.lang.Thread.run(Thread.java:923)
09:07:31.586  A  runtime.cc:686] Runtime aborting...
                 runtime.cc:686] Dumping all threads without mutator lock held
                 runtime.cc:686] All threads:
                 runtime.cc:686] DALVIK THREADS (73):
                 runtime.cc:686] "CHIP Device Controller IO Thread" prio=5 tid=68 Runnable
                 runtime.cc:686]   | group="" sCount=0 dsCount=0 flags=0 obj=0x13300030 self=0xb4000073e95f2540
                 runtime.cc:686]   | sysTid=16455 nice=0 cgrp=default sched=0/0 handle=0x71ce3f7cc0
                 runtime.cc:686]   | state=R schedstat=( 466208276 31765100 259 ) utm=37 stm=9 core=6 HZ=100
                 runtime.cc:686]   | stack=0x71ce300000-0x71ce302000 stackSize=995KB
                 runtime.cc:686]   | held mutexes= "abort lock" "mutator lock"(shared held)
                 runtime.cc:686]   native: #00 pc 00000000004a07d0  /apex/com.android.art/lib64/libart.so (art::DumpNativeStack(std::__1::basic_ostream<char, std::__1::char_traits<char> >&, int, BacktraceMap*, char const*, art::ArtMethod*, void*, bool)+140)
                 runtime.cc:686]   native: #01 pc 00000000005ada14  /apex/com.android.art/lib64/libart.so (art::Thread::DumpStack(std::__1::basic_ostream<char, std::__1::char_traits<char> >&, bool, BacktraceMap*, bool) const+376)
                 runtime.cc:686]   native: #02 pc 00000000005cab4c  /apex/com.android.art/lib64/libart.so (art::DumpCheckpoint::Run(art::Thread*)+924)
                 runtime.cc:686]   native: #03 pc 00000000005c4a8c  /apex/com.android.art/lib64/libart.so (art::ThreadList::RunCheckpoint(art::Closure*, art::Closure*)+528)
                 runtime.cc:686]   native: #04 pc 00000000005c3c58  /apex/com.android.art/lib64/libart.so (art::ThreadList::Dump(std::__1::basic_ostream<char, std::__1::char_traits<char> >&, bool)+1920)
                 runtime.cc:686]   native: #05 pc 000000000055e858  /apex/com.android.art/lib64/libart.so (art::Runtime::Abort(char const*)+1864)
                 runtime.cc:686]   native: #06 pc 0000000000013978  /system/lib64/libbase.so (android::base::SetAborter(std::__1::function<void (char const*)>&&)::$_3::__invoke(char const*)+76)
                 runtime.cc:686]   native: #07 pc 0000000000012fa4  /system/lib64/libbase.so (android::base::LogMessage::~LogMessage()+320)
                 runtime.cc:686]   native: #08 pc 0000000000385ff4  /apex/com.android.art/lib64/libart.so (art::JavaVMExt::JniAbort(char const*, char const*)+2572)
                 runtime.cc:686]   native: #09 pc 000000000038606c  /apex/com.android.art/lib64/libart.so (art::JavaVMExt::JniAbortV(char const*, char const*, std::__va_list)+108)
                 runtime.cc:686]   native: #10 pc 0000000000377d50  /apex/com.android.art/lib64/libart.so (art::(anonymous namespace)::ScopedCheck::AbortF(char const*, ...)+144)
                 runtime.cc:686]   native: #11 pc 0000000000376868  /apex/com.android.art/lib64/libart.so (art::(anonymous namespace)::ScopedCheck::CheckPossibleHeapValue(art::ScopedObjectAccess&, char, art::(anonymous namespace)::JniValueType)+3108)
                 runtime.cc:686]   native: #12 pc 00000000003754c8  /apex/com.android.art/lib64/libart.so (art::(anonymous namespace)::ScopedCheck::Check(art::ScopedObjectAccess&, bool, char const*, art::(anonymous namespace)::JniValueType*)+612)
                 runtime.cc:686]   native: #13 pc 000000000036b858  /apex/com.android.art/lib64/libart.so (art::(anonymous namespace)::CheckJNI::NewStringUTF(_JNIEnv*, char const*)+696)
[... cut ...]
09:07:31.740  A        #29 pc 00000000024258b8  /data/app/~~Bdm2OOjoieg1UPsxe101Tg==/dev.atlantis.bhtronics_dev-zVQTHxePdrFy7-1Qy5--Kw==/base.apk (offset 0xc2f000)
09:07:31.740  A        #30 pc 0000000002505374  /data/app/~~Bdm2OOjoieg1UPsxe101Tg==/dev.atlantis.bhtronics_dev-zVQTHxePdrFy7-1Qy5--Kw==/base.apk (offset 0xc2f000)
09:07:31.740  A        #31 pc 0000000000d28330  /data/app/~~Bdm2OOjoieg1UPsxe101Tg==/dev.atlantis.bhtronics_dev-zVQTHxePdrFy7-1Qy5--Kw==/base.apk (offset 0xc2f000)
09:07:31.740  A        #32 pc 0000000000d19410  /data/app/~~Bdm2OOjoieg1UPsxe101Tg==/dev.atlantis.bhtronics_dev-zVQTHxePdrFy7-1Qy5--Kw==/base.apk (offset 0xc2f000)

yunhanw-google, 2024-01-14:
sounds like we need to catch exception in OnError function in android chip-tool.

  private val reportCallback =
    object : ReportCallback {
      override fun onError(
        attributePath: ChipAttributePath?,
        eventPath: ChipEventPath?,
        ex: Exception
      ) {
        if (attributePath != null) {
          Log.e(TAG, "Report error for $attributePath: $ex")
        }
        if (eventPath != null) {
          Log.e(TAG, "Report error for $eventPath: $ex")
        }
      }

https://github.com/project-chip/connectedhomeip/blob/4676d277af113d6e07e0de235c8ca8b70237109a/examples/android/CHIPTool/app/src/main/java/com/google/chip/chiptool/clusterclient/WildcardFragment.kt#L67

https://github.com/project-chip/connectedhomeip/blob/4676d277af113d6e07e0de235c8ca8b70237109a/examples/android/CHIPTool/app/src/main/java/com/google/chip/chiptool/clusterclient/MultiAdminClientFragment.kt#L212

https://github.com/project-chip/connectedhomeip/blob/4676d277af113d6e07e0de235c8ca8b70237109a/examples/java-matter-controller/java/src/com/matter/controller/commands/pairing/PairOnNetworkLongImSubscribeCommand.kt#L33

pgatti86, 2024-01-15:
@yunhanw-google I think that exceptions need to be exposed to java/kotlin.
At the moment I cannot catch any exeption. Is it correct? Thanks

yunhanw-google, 2024-01-17:
@pgatti86  could you try this fix? thanks

yunhanw-google, 2024-01-17:
@pgatti86 I tried android chip-tool and do the successful pairing, then power-off the accessory, do read for basic info cluster, the exchange manager retries counter 4, then onConnectionFailure is called. It seems it is working correctly.

01-17 15:05:13.203 29951 29999 D EM      : Retransmitting MessageCounter:137035299 on exchange 21526i Send Cnt 4
..
01-17 15:05:48.353 29951 29999 E SC      : CASESession timed out while waiting for a response from the peer. Current state was 4
01-17 15:05:48.355 29951 29999 D IN      : SecureSession[0xb4000077e60ca690]: Released - Type:2 LSID:24932
01-17 15:05:48.355 29951 29999 D DIS     : OperationalSessionSetup[1:0000000000000001]: State change 4 --> 2
01-17 15:05:48.355 29951 29999 D DIS     : OperationalSessionSetup[1:0000000000000001]: State change 2 --> 1
01-17 15:05:48.366 29951 29999 E ChipClient: Unable to get connected device with nodeId 1
01-17 15:05:48.366 29951 29999 E ChipClient: chip.devicecontroller.ConnectionFailureException: src/protocols/secure_channel/CASESession.cpp:554: CHIP Error 0x00000032: Timeout

yunhanw-google, 2024-01-17:
@pgatti86 From your log, it seems the code goes from https://github.com/project-chip/connectedhomeip/blob/f2fa44a2fe5880355ae6c53b03d38e3758c00ced/src/app/OperationalSessionSetup.cpp#L245, I guess it is possible that it return earlier when it tries the last retransmission so that it cannot call DequeueConnectionCallbacks.

If you can consistently reproduce this issue, and put additional logging and find the actual error point, then we can put up the right fix on it.

thanks

pgatti86, 2024-01-18:
I confirm that I can consistently reproduce this issue.
For example I try to write the Occupied Heating setpoint after turning off my device.

Here the exception details:

Failed to Send CHIP MessageCounter:77699762 on exchange 30421i sendCount: 4 max retries: 4
Time out! failed to receive write response from Exchange: 30420i
writeOccupiedHeatingSetpointAttribute failure
chip.devicecontroller.ChipClusterException: CHIP cluster error: 1
Time out! failed to receive report data from Exchange: 30421i
java_vm_ext.cc:591] JNI DETECTED ERROR IN APPLICATION: JNI GetObjectClass called with pending exception java.lang.NullPointerException:  java_vm_ext.cc:591] (Throwable with no stack trace)
                                                                                                    java_vm_ext.cc:591] 
                                                                                                    java_vm_ext.cc:591]     in call to GetObjectClass 
                                                                                                    runtime.cc:691] Runtime aborting...
                                                                                                    runtime.cc:691] Dumping all threads without mutator lock held

yunhanw-google, 2024-01-19:
@pgatti86 what is the commit Id are you using to reproduce this issue? I am using latest in master, and still cannot reproduce it. I am just using android-chip-tool, power off my switch, then write the label using basic cluster fragment in UI.

pgatti86, 2024-01-22:
@yunhanw-google I have checked out the tagged version 1.2.0.1

pgatti86, 2024-01-22:
@yunhanw-google After your suggestion I have checked out the master branch. Same result. When I write an attribute with my device turned off an exception is thrown. 

Here some logs:

Failed to Send CHIP MessageCounter:70053300 on exchange 29799i sendCount: 4 max retries: 4
Time out! failed to receive write response from Exchange: 29799i
WriteAttributesCallback::ReportError is called with 50
writeOccupiedHeatingSetpointAttribute failure
chip.devicecontroller.ChipDeviceControllerException: src/app/WriteClient.cpp:504: CHIP Error 0x00000032: Timeout
ava_vm_ext.cc:591] JNI DETECTED ERROR IN APPLICATION: JNI GetObjectClass called with pending exception java.lang.NullPointerException: 
                                                      java_vm_ext.cc:591] (Throwable with no stack trace)
                                                      java_vm_ext.cc:591]
                                                      java_vm_ext.cc:591]     in call to GetObjectClass
                                                      runtime.cc:691] Runtime aborting...
                                                      runtime.cc:691] Dumping all threads without mutator lock held
                                                      runtime.cc:691] All threads:
                                                      runtime.cc:691] DALVIK THREADS (69):
                                                      runtime.cc:691] "main" prio=10 tid=1 Native
                                                      
Is there anything else I can do to help you?

yunhanw-google, 2024-01-24:
@pgatti86 are you using android chip-tool, if not, can you trying the latest android chip-tool, if still reproduced, please let me the detailed step. thanks

joonhaengHeo, 2024-01-25:
@pgatti86 
Please share full log when this issue is reproduced.

yunhanw-google, 2024-01-25:
@joonhaengHeo  and I both tried several times using master branch and v1.2.0.1 branch, we both cannot reproduce it. Maybe you can try android chip-tool using your scenario, if that works, then maybe something wrong on your application? thanks @pgatti86

yunhanw-google, 2024-01-31:
@pgatti86 any update on it? thanks

pgatti86, 2024-02-13:
@yunhanw-google 
We have found the reason behind the crash. We are using suspendCancellableCoroutine for bridging callback operations with suspended functions. If we resume with an exception in case of error we get the behavior described. Obviously the generated exception was catched but it doesn't seem to be enough and the stacktrace generated doesn't help neither.

```
suspendCancellableCoroutine { continuation ->
                thermostatCluster.writeTemperatureSetpointHoldDurationAttribute(object :
                    DefaultClusterCallback {
                    override fun onSuccess() {
                        continuation.resume(Unit)
                    }

                    override fun onError(ex: Exception) {
                        Log.e(TAG, "writeTemperatureSetpointHoldDurationAttribute failure", ex)
                        continuation.resumeWithException(ex.cause!!)
                    }
                }, value)
            }
```

yunhanw-google, 2024-02-15:
@pgatti86  thanks for the reply. May i know which exception is captured, and further lead to the above issue? How do you fix this issue?

pgatti86, 2024-02-21:
@yunhanw-google the issue was fixed resuming the continuation without the exception, even into the onError callback.
This requires resuming the continuation with a null or invalid value in case of error.

yunhanw-google, 2024-02-22:
I see, thanks for clarification.

closed 2024-02-22 by yunhanw-google