project-chip/connectedhomeip #30634: [Build] It looks like this version of Python does not support SSL.

koenlee93, 2023-11-24:
### Build issue(s)

ubuntu@ubuntu:~/connectedhomeip$ source scripts/activate.sh

  WELCOME TO...

         █
         █
     ▄   █   ▄                                █     █
     ▀▀█████▀▀      ▄▀▀▀▄ ▄▀▀▀▄    ▄▀▀▀▀▄█  ▀▀█▀▀▀▀▀█▀▀   ▄▀▀▀▀▄    ▄▀▀
   ▀█▄       ▄█▀   █     █     █  █      █    █     █    █▄▄▄▄▄▄█  █
     ▀█▄   ▄█▀     █     █     █  █      █    █     █    █         █
  ▄██▀▀█   █▀▀██▄  █     █     █   ▀▄▄▄▄▀█    ▀▄▄   ▀▄▄   ▀▄▄▄▄▀   █
 ▀▀    █   █    ▀▀

  BOOTSTRAP! Bootstrap may take a few minutes; please be patient.

Downloading and installing packages into local source directory:

  Setting up CIPD package manager...[|]======================================================================
It looks like this version of Python does not support SSL. This is common
when using Homebrew. If using Homebrew please run the following commands.
If not using Homebrew check how your version of Python was built.

brew install openssl  # Probably already installed, but good to confirm.
brew uninstall python && brew install python
======================================================================                                                                                                                                                               Traceback (most recent call last):
  File "/home/ubuntu/connectedhomeip/third_party/pigweed/repo/pw_env_setup/py/pw_env_setup/env_setup.py", line 982, in <module>
    sys.exit(main())
  File "/home/ubuntu/connectedhomeip/third_party/pigweed/repo/pw_env_setup/py/pw_env_setup/env_setup.py", line 974, in main
    return EnvSetup(**vars(parse())).setup()
  File "/home/ubuntu/connectedhomeip/third_party/pigweed/repo/pw_env_setup/py/pw_env_setup/env_setup.py", line 573, in setup
    result = step(spin)
  File "/home/ubuntu/connectedhomeip/third_party/pigweed/repo/pw_env_setup/py/pw_env_setup/env_setup.py", line 667, in cipd
    cipd_client = cipd_wrapper.init(
  File "/home/ubuntu/connectedhomeip/third_party/pigweed/repo/pw_env_setup/py/pw_env_setup/cipd_setup/wrapper.py", line 378, in init
    bootstrap(client, silent, rosetta=rosetta)
  File "/home/ubuntu/connectedhomeip/third_party/pigweed/repo/pw_env_setup/py/pw_env_setup/cipd_setup/wrapper.py", line 328, in bootstrap
    tmp.write(client_bytes(rosetta))
  File "/home/ubuntu/connectedhomeip/third_party/pigweed/repo/pw_env_setup/py/pw_env_setup/cipd_setup/wrapper.py", line 228, in client_bytes
    conn = https_connect_with_proxy(CIPD_HOST)
  File "/home/ubuntu/connectedhomeip/third_party/pigweed/repo/pw_env_setup/py/pw_env_setup/cipd_setup/wrapper.py", line 201, in https_connect_with_proxy
    conn = httplib.HTTPSConnection(url.hostname, url.port)
  File "/usr/lib/python3.10/http/client.py", line 1410, in __init__
    super(HTTPSConnection, self).__init__(host, port, timeout,
  File "/usr/lib/python3.10/http/client.py", line 852, in __init__
    (self.host, self.port) = self._get_hostport(host, port)
  File "/usr/lib/python3.10/http/client.py", line 886, in _get_hostport
    i = host.rfind(':')
AttributeError: 'NoneType' object has no attribute 'rfind'
Installing pip requirements for all...

### Anything else?

[image]

koenlee93, 2023-11-24:
What may be the cause of this problem

bzbarsky-apple, 2023-11-28:
@koenlee93 The cause is that the Python you have installed does not support SSL, no?

koenlee93, 2023-11-28:
你好!我是李星宇，已收到你发的邮件，我会尽快处理的。