project-chip/connectedhomeip #30640: [1.2] Build issue

WeeXnes, 2023-11-24:
### Reproduction steps

im trying to build the project, i did eveything according to the build guide for linux. but eveytime i try to run the 
```
ninja -C out/host check
```
i get the following output;
```
╭╴at …/connectedhomeip on  mast… () via  v3.8.10 (pigweed-venv) took 10m 
╰─ ninja -C out/host check
ninja: Entering directory `out/host'
[733/851] ACTION //src/platform/tests:TestPlatformMgr.run(//build/toolchain/linux:linux_x64_gcc)
FAILED: gen/src/platform/tests/TestPlatformMgr.run.pw_pystamp 
python3 ../../third_party/pigweed/repo/pw_build/py/pw_build/python_runner.py --gn-root ../../ --current-path ../../src/platform/tests --default-toolchain=//build/toolchain/linux:linux_x64_gcc --current-toolchain=//build/toolchain/linux:linux_x64_gcc --touch gen/src/platform/tests/TestPlatformMgr.run.pw_pystamp --capture-output --module pw_unit_test.test_runner --python-virtualenv-config python/gen/matter_build_venv/venv_metadata.json --python-dep-list-files gen/src/platform/tests/TestPlatformMgr.run_metadata_path_list.txt -- --runner ../../third_party/pigweed/repo/targets/host/run_test --test tests/TestPlatformMgr
ERR ../../third_party/pigweed/repo/targets/host/run_test exited with status 134
OUT [Pid: 30545]
'#0:','PlatformMgr tests'
'#2:','Setup                                                ','PASSED'
[1700840479.627644][30548:30548] CHIP:DL: ChipLinuxStorage::Init: Using KVS config file: /tmp/chip_kvs
[1700840479.628125][30548:30548] CHIP:DL: ChipLinuxStorage::Init: Using KVS config file: /tmp/chip_factory.ini
[1700840479.628225][30548:30548] CHIP:DL: ChipLinuxStorage::Init: Using KVS config file: /tmp/chip_config.ini
[1700840479.628269][30548:30548] CHIP:DL: ChipLinuxStorage::Init: Using KVS config file: /tmp/chip_counters.ini
[1700840479.628510][30548:30548] CHIP:DL: writing settings to file (/tmp/chip_counters.ini-GyMSt4)
[1700840479.628837][30548:30548] CHIP:DL: renamed tmp file to file (/tmp/chip_counters.ini)
[1700840479.628859][30548:30548] CHIP:DL: NVS set: chip-counters/reboot-count = 19 (0x13)
[1700840479.628952][30548:30548] CHIP:DL: writing settings to file (/tmp/chip_config.ini-3pXlO4)
[1700840479.629134][30548:30548] CHIP:DL: renamed tmp file to file (/tmp/chip_config.ini)
[1700840479.629151][30548:30548] CHIP:DL: NVS set: chip-config/regulatory-location = 0 (0x0)
[1700840479.629251][30548:30548] CHIP:DL: writing settings to file (/tmp/chip_config.ini-TopE23)
[1700840479.637514][30548:30548] CHIP:DL: renamed tmp file to file (/tmp/chip_config.ini)
[1700840479.637594][30548:30548] CHIP:DL: NVS set: chip-config/location-capability = 2 (0x2)
[1700840479.638028][30548:30548] CHIP:DL: Failed to get Ethernet interface
[1700840479.638250][30548:30548] CHIP:DL: Failed to reset Ethernet statistic counts
[1700840479.638383][30548:30548] CHIP:DL: Got WiFi interface: wlo1
[1700840479.639536][30548:30548] CHIP:DL: Found the primary WiFi interface:wlo1
[1700840479.639738][30548:30548] CHIP:DL: writing settings to file (/tmp/chip_counters.ini-578wO6)
[1700840479.640040][30548:30548] CHIP:DL: renamed tmp file to file (/tmp/chip_counters.ini)
[1700840479.640065][30548:30548] CHIP:DL: NVS set: chip-counters/total-operational-hours = 0 (0x0)
[1700840479.640078][30548:30548] CHIP:DL: Inet Layer shutdown
[1700840479.640087][30548:30548] CHIP:DL: BLE shutdown
[1700840479.640098][30548:30548] CHIP:DL: System Layer shutdown
'#3:','Test PlatformMgr::Init/Shutdown                      ','PASSED'
[1700840479.640561][30548:30548] CHIP:DL: ChipLinuxStorage::Init: Using KVS config file: /tmp/chip_kvs
[1700840479.640578][30548:30548] CHIP:DL: ChipLinuxStorage::Init: Attempt to re-initialize with KVS config file: /tmp/chip_kvs
[1700840479.640642][30548:30548] CHIP:DL: ChipLinuxStorage::Init: Using KVS config file: /tmp/chip_factory.ini
[1700840479.640649][30548:30548] CHIP:DL: ChipLinuxStorage::Init: Attempt to re-initialize with KVS config file: /tmp/chip_factory.ini
[1700840479.640657][30548:30548] CHIP:DL: ChipLinuxStorage::Init: Using KVS config file: /tmp/chip_config.ini
[1700840479.640663][30548:30548] CHIP:DL: ChipLinuxStorage::Init: Attempt to re-initialize with KVS config file: /tmp/chip_config.ini
[1700840479.640671][30548:30548] CHIP:DL: ChipLinuxStorage::Init: Using KVS config file: /tmp/chip_counters.ini
[1700840479.640677][30548:30548] CHIP:DL: ChipLinuxStorage::Init: Attempt to re-initialize with KVS config file: /tmp/chip_counters.ini
[1700840479.640813][30548:30548] CHIP:DL: writing settings to file (/tmp/chip_counters.ini-fmuAJ4)
[1700840479.641064][30548:30548] CHIP:DL: renamed tmp file to file (/tmp/chip_counters.ini)
[1700840479.641087][30548:30548] CHIP:DL: NVS set: chip-counters/reboot-count = 20 (0x14)
[1700840479.641464][30548:30548] CHIP:DL: Failed to get Ethernet interface
[1700840479.641701][30548:30548] CHIP:DL: Failed to reset Ethernet statistic counts
[1700840479.641860][30548:30548] CHIP:DL: Got WiFi interface: wlo1
[1700840479.642912][30548:30548] CHIP:DL: Found the primary WiFi interface:wlo1
[1700840479.643084][30548:30553] CHIP:DL: CHIP task running
[1700840479.647236][30548:30552] CHIP:DL: TRACE: Bus acquired for name 
[1700840479.647412][30548:30552] CHIP:DL: CREATE service object at /chipoble/7754/service
[1700840479.649642][30548:30552] CHIP:DL: Create characteristic object at /chipoble/7754/service/c1
[1700840479.649945][30548:30552] CHIP:DL: Create characteristic object at /chipoble/7754/service/c2
[1700840479.650026][30548:30552] CHIP:DL: CHIP BTP C1 /chipoble/7754/service
[1700840479.650035][30548:30552] CHIP:DL: CHIP BTP C2 /chipoble/7754/service
[1700840479.650043][30548:30552] CHIP:DL: CHIP_ENABLE_ADDITIONAL_DATA_ADVERTISING is FALSE
[1700840479.656608][30548:30553] CHIP:DL: BlueZ integration init success
[1700840479.662486][30548:30552] CHIP:DL: BluezPeripheralRegisterAppDone done
[1700840479.667789][30548:30556] CHIP:DL: CHIP task running
[1700840479.667891][30548:30556] CHIP:DL: HandlePlatformSpecificBLEEvent 16392
[1700840479.668016][30548:30556] CHIP:SPT: VerifyOrDie failure at src/platform/CommissionableDataProvider.cpp:32: gCommissionableDataProvider != nullptr
Aborted
INF Test 1/1: [FAIL] TestPlatformMgr in 0.061 s
[750/851] ACTION //src/messaging/tests:TestReliableMessageProtocol.run(//build/toolchain/linux:linux_x64_gcc)
ninja: build stopped: subcommand failed.
```

### Bug prevalence

when trying to setup build env

### GitHub hash of the SDK that was being used

b02f7744206f158a0eb9d074c3769e0bc3323e4d

### Platform

other

### Platform Version(s)

1.2

### Type

Platform Issue

trantrieuphi, 2023-11-28:
I also encountered the same situation and could not resolve it. I don't know if you have solved this problem yet?

bzbarsky-apple, 2023-11-28:
@WeeXnes Is this a problem on tip too?  Or just on https://github.com/project-chip/connectedhomeip/commit/b02f7744206f158a0eb9d074c3769e0bc3323e4d ?

WeeXnes, 2023-11-30:
> @WeeXnes Is this a problem on tip too?  Or just on https://github.com/project-chip/connectedhomeip/commit/b02f7744206f158a0eb9d074c3769e0bc3323e4d ?

At the time of posting this issue it was the latest commit, so i dont know wether it is just an issue on that commit or on tip as well, i dont have the time to fully clone a 20Gb+ repo every commit to check if the issue has been resolved, thats why i opened this issue.

WeeXnes, 2023-11-30:
> I also encountered the same situation and could not resolve it. I don't know if you have solved this problem yet?
> 

Sadly not, but i got it working another way, since i wanted to build for esp32, i followed the Espressif guide for the esp matter sdk, which is built on top of this repo, and it worked perfectly fine

PR #31052 (2023-12-18): Fix TestPlatformMgr crash
- TestPlatformMgr did not set a CommissionableDataProvider.
- Recent code changes make that test hit codepaths that try to setup BLE advertisement on Linux platform when default-built.
- This PR adds a CommissionableDataProvider instance for the whole test with default test values. Tests now pass.

Fixes #30640
commit 44e1a708 (2023-12-18): Fix TestPlatformMgr crash (#31052)

- TestPlatformMgr did not set a CommissionableDataProvider.
- Recent code changes make that test hit codepaths that try
  to setup BLE advertisement on Linux platform when default-built.
- This PR adds a CommissionableDataProvider instance for the whole
  test with default test values. Tests now pass.

Fixes #30640
closed 2023-12-18 by mergify[bot]