project-chip/connectedhomeip #30702: [BUG] Regression caused by #30071 - android tv-app crashes

chrisdecenzo, 2023-11-29:
### Reproduction steps

tv-app's platform-app.apk crashes when it starts, see:

https://csamembers.slack.com/archives/C05KATCH551/p1701223521545509

### Bug prevalence

always

### GitHub hash of the SDK that was being used

TOT

### Platform

android

sharadb-amazon, 2023-11-30:
This also leads the Android tv-casting-app to crash: 

```
11-30 14:10:15.475 22031 22031 F DEBUG   : pid: 21996, tid: 21996, name: om.chip.casting  >>> com.chip.casting <<<
11-30 14:10:15.475 22031 22031 F DEBUG   : uid: 10399
11-30 14:10:15.475 22031 22031 F DEBUG   : tagged_addr_ctrl: 0000000000000001 (PR_TAGGED_ADDR_ENABLE)
11-30 14:10:15.475 22031 22031 F DEBUG   : signal 6 (SIGABRT), code -1 (SI_QUEUE), fault addr --------
11-30 14:10:15.475 22031 22031 F DEBUG   : Abort message: 'JNI DETECTED ERROR IN APPLICATION: JNI FindClass called with pending exception java.lang.ClassNotFoundException: 
Didn't find class "chip.devicecontroller.ControllerParams" on path: DexPathList[[zip file "/data/app/~~ViTuhCE9G9ofJtTT4KmYPA==/com.chip.casting-b2f0Y3A9ATatDOfqzOoUsg==/base.apk"],
nativeLibraryDirectories=[/data/app/~~ViTuhCE9G9ofJtTT4KmYPA==/com.chip.casting-b2f0Y3A9ATatDOfqzOoUsg==/lib/arm64, /data/app/~~ViTuhCE9G9ofJtTT4KmYPA==/com.chip.casting-b2f0Y3A9ATatDOfqzOoUsg==/base.apk!/lib/arm64-v8a, /system/lib64, /system_ext/lib64]]
```

Jaehoon-You, 2023-12-01:
Hello. Android virtual-device-app is also not a commissioner

> 2023-12-01 15:26:38.340 28529-28529 DEBUG                               pid-28529                            A  pid: 28379, tid: 28379, name: tual.device.app  >>> com.matter.virtual.device.app <<<
2023-12-01 15:26:38.340 28529-28529 DEBUG                               pid-28529                            A  Abort message: 'JNI DETECTED ERROR IN APPLICATION: JNI FindClass called with pending exception java.lang.ClassNotFoundException: Didn't find class "chip.devicecontroller.ControllerParams" on path: DexPathList[[zip file "/data/app/~~0pEvPxKRMou7DUx-yyV7kw==/com.matter.virtual.device.app-LtvFSnciupWa4z-Zi9zZ8A==/base.apk"],nativeLibraryDirectories=[/data/app/~~0pEvPxKRMou7DUx-yyV7kw==/com.matter.virtual.device.app-LtvFSnciupWa4z-Zi9zZ8A==/lib/arm64, /data/app/~~0pEvPxKRMou7DUx-yyV7kw==/com.matter.virtual.device.app-LtvFSnciupWa4z-Zi9zZ8A==/base.apk!/lib/arm64-v8a, /system/lib64, /system_ext/lib64]]

PR #30769 (2023-12-01): Revert "Update Java8 Optional handling in JniReferences (#30071)"
This reverts commit 4e438d69c63d259dafdc3fc3bb1748d7de27e044.

https://github.com/project-chip/connectedhomeip/pull/30071 employs a randomly selected class to assess if the Java code supports Java 8 properly. The chosen class and method, sourced from src/controller/java, are arbitrary. However, it's important to note that all classes in this package are exclusively for controller functionalities. For commissionable devices, the implementation is confined to src/platform/android/java/chip/platform. Therefore, this approach may no [... cut]
closed 2023-12-01 by mergify[bot]