project-chip/connectedhomeip #30707: [Feature] : Need empty project scaffolding

antonio-gabriele, 2023-11-29:
### Feature description

I need an empty project scaffolding on top of which I can implement my own device, without use examples/..app../..platform.. folder structure. Or please give documentation about use connectedhomeip as static/shared library and related headers. Also, please consider using different build system than Ninja (as Cmake for example).

### Platform

all

Damian-Nordic, 2023-11-29:
fyi: there are several CMake-based platforms in Matter that integrate Matter's build system (based on GN) using CMake's `ExternalProject` feature. You may take a look at `config/common/cmake` directory that contains CMake helper functions for exactly that.

antonio-gabriele, 2023-11-30:
Do you have any exmple?

Damian-Nordic, 2023-11-30:
`mbed`, `nrfconnect, `telink` and `openiot` platforms use it. Search for `matter_build(` in the repository and you will find the usage examples.