project-chip/connectedhomeip #30746: [CI/CD] detect and prevent unintended third_party repo updates

andy31415, 2023-11-30:
Due to codegen, developers often do `git commit -a` to add all changed files, however this also can easily unintentionally change submodues if a they are not synchronized.

Then we have to undo that change (when/if we notice) like in

https://github.com/project-chip/connectedhomeip/pull/30451
https://github.com/project-chip/connectedhomeip/pull/28164

and others. These were the ones that made through, code reviews catch others some of the time, however it is tedious and humans make mistakes.

We should have something that flags PRs that intentionally change submodules and flak (likely failing CI) if unintentional submodules change. Something like a `Changing submodules: lwip, gecko_sdk` in a comment (I like this because specific) or a label (like `changing-submodules-on-purpose`) should probably be a requirement to have PRs go in that update submodules.

We also have to make sure dependabot changes go through (configured through https://github.com/project-chip/connectedhomeip/blob/master/.github/dependabot.yml, they create PRs like https://github.com/project-chip/connectedhomeip/pulls?q=is%3Apr+author%3Aapp%2Fdependabot+is%3Aclosed)

vatsalghelani-csa, 2024-04-11:
Either we need a script to detect changes or maybe some rule based on paths changed ... if anything in "/third_party/" is modified we need to run some validation.

https://github.com/project-chip/connectedhomeip/blob/master/.github/workflows/kotlin-style.yaml#L6  -  is an example of a workflow that runs for specific path changes only. So we need some logic like:
run this workflow when anything under third_party changes (maybe? or maybe .gitmodules is the right path?)
when it runs, check that some specific label is applied to the PR - this needs figuring out how.
skip running it for dependabot changes

kliao-csa, 2024-04-18:
I think what we could try for starters is create a yaml that:

1. Runs on Pull Request only
2. Checks specifically for /third_party/ and .gitmodules, so changes to both will trigger it.
3. contains( github.event.pull_request.labels.*.name, 'My Label') allows you to check against labels
4. if: ${{ github.actor != 'dependabot[bot]' }} allows you to exclude dependably

I can put together an initial script for these requirements this week and have you look over it, Vatsal.

kliao-csa, 2024-04-18:
To implement the `Changing submodules: lwip, gecko_sdk` suggestion, we may use `if: contains(github.event.comment.body, 'Changing submodules'`. However this institutes a somewhat less intuitive rule that would require exact text input in a comment, so it may be more trouble than it's worth. If we wanted to make sure the submodules listed in the comments actually match the submodule paths changed 1:1, we'd (for the cleanest implementation) need a custom Python script instead of just this workflow. Let's consider how often this kind of issue occurs before going down that path.

closed 2024-04-19 by kliao-csa