project-chip/connectedhomeip #30749: [CI/CD] Add extra lint/validators for XML clusters

andy31415, 2023-11-30:
We currently do not validate XML for clusters at all.

In theory we have https://github.com/project-chip/zap/blob/master/zcl-builtin/shared/schema/zcl.xsd that can validate the schema. Should figure out how zap-cli could be used to validate this or if not, use the xsd from the zap repo itself (even though that may be a version mismatch).

We also likely want some extra validation for some common typos (like forcing casing on things), to avoid PRs like https://github.com/project-chip/connectedhomeip/pull/30740 and https://github.com/project-chip/connectedhomeip/pull/29706 needing to be done over and over again.

This may need splitting into separate tasks:
   - add validation for xsd (and this will find issues that need fixing)
   - add casing enforcement for some standard types (either in matter or in xml files)

andy31415, 2023-12-14:
we should catch typos such as found by #31015

cecille, 2025-04-29:
Closing this in favour of the alchemy issue

closed 2025-04-29 by cecille