project-chip/connectedhomeip #30753: [1.2] [Scenes] RemoveFabric command leaves a nonzero SceneCount

zacwbond, 2023-11-30:
### Reproduction steps

I have a device that supports Groups and Scenes on endpoint 2.  The `EP2.Scenes.SceneCount` attribute is 0 and the `EP2.Scenes.RemainingCapacity` attribute is 7 after a factory reset.   

I assume that the DUT is commissioned onto a single fabric.

1. Add one group and two scenes.  The easiest way is to just follow the steps for the Scenes test TC-S-2.2 up until step 2b, and then do one more AddScene command.
2. Check the EP2.Scenes.SceneCount attribute and see that it reports 2.
3. Check the EP2.Scenes.RemainingCapacity attribute and see that it reports 5.
4. Send the RemoveFabric command to `EP0.NodeOperationalCredentials` for the current fabric index.
5. Re-commission the device onto a new fabric.
6. Check the `EP2.Scenes.SceneCount` attribute and see that it still reports 2 instead of 0.
7. Check the `EP2.Scenes.RemainingCapacity` attribute and see that it reports 7.

I would expect the`EP2.Scenes.SceneCount` would return to 0 Based on 1.4.6 of the application clusters spec:
> Upon leaving a fabric with the RemoveFabric command of the Operational Credentials Cluster, all scenes data for the associated fabric SHALL be removed from the Scene Table.

Note that if you send a `RemoveAllGroups` command, then `EP2.Scenes.SceneCount` will return to 0.  The same happens for `RemoveAllScenes`.  What I would expect is that a `RemoveFabricCommand` would trigger `RemoveAllGroups` for that fabric, which would then trigger `RemoveAllScenes`, but that doesn't seem to be happening.  

Maybe I am misunderstanding something here, but there seems to be no way to get rid of the two scenes associated with the deleted fabric and group besides a factory reset.  

To help see what the program flow is, I added some tracing to the logs related to group and scene removal, and I also added a trace showing writes and deletes of persisted data keys.  

Here's a log for group removal:
```
I (319438) chip[DL]: -- GroupWillBeRemoved()
I (319442) chip[DL]: -- DeleteAllScenesInGroup(1)
I (319446) chip[DL]: -- RemoveScene(0)
I (319449) chip[DL]: KVM -- Put "g/scc/e/2"
I (319459) chip[DL]: KVM -- Put "f/2/e/2/sc"
I (319462) chip[DL]: KVM -- Delete "f/2/e/2/sc/0"
I (319474) chip[DL]: --RemoveScene(1)
I (319477) chip[DL]: KVM -- Put "g/scc/e/2"
I (319482) chip[DL]: KVM -- Put "f/2/e/2/sc"
I (319485) chip[DL]: KVM -- Delete "f/2/e/2/sc/1"
I (319511) chip[DL]: -- GroupWillBeRemoved()
I (319512) chip[DL]: -- DeleteAllScenesInGroup(0)
I (319515) chip[DL]: -- RemoveScene(0)
I (319515) chip[DL]: -- RemoveScene(1)
I (319515) chip[DL]: -- RemoveScene(2)
I (319515) chip[DL]: -- RemoveScene(3)
I (319516) chip[DL]: -- RemoveScene(4)
I (319516) chip[DL]: -- RemoveScene(5)
I (319517) chip[DL]: -- RemoveScene(6)
I (319524) chip[DL]: KVM -- Delete "f/2/g/1/e/2"
I (319545) chip[DL]: KVM -- Delete "f/2/g/1"
I (319557) chip[DL]: KVM -- Put "f/2/g"
```

Here is a similar log for RemoveFabric().  It looks like the group is deleted, but none of the scenes associated with the group are deleted.

```
I (2445277) chip[ZCL]: OpCreds: Received a RemoveFabric Command for FabricIndex 0x2
I (2445281) chip[DL]: KVM -- Delete "f/2/m"
I (2445294) chip[DL]: KVM -- Delete "f/2/o"
I (2445310) chip[DL]: KVM -- Delete "f/2/n"
I (2445316) chip[DL]: KVM -- Delete "f/2/i"
I (2445323) chip[DL]: KVM -- Delete "f/2/r"
I (2445328) chip[DL]: KVM -- Put "g/fidx"
I (2445330) chip[FP]: Fabric (0x2) deleted.
I (2445331) chip[ZCL]: OpCreds: Fabric index 0x2 was removed
I (2445342) chip[DL]: KVM -- Delete "g/sum"
I (2445360) chip[DL]: KVM -- Delete "f/2/gk/2"
I (2445368) chip[DL]: KVM -- Put "f/2/gk/1"
I (2445374) chip[DL]: KVM -- Put "f/2/g"
I (2445383) chip[DL]: KVM -- Delete "f/2/gk/1"
I (2445402) chip[DL]: KVM -- Put "f/2/g"
I (2445410) chip[DL]: KVM -- Delete "f/2/g/1/e/2"
I (2445417) chip[DL]: KVM -- Delete "f/2/g/1"
I (2445426) chip[DL]: KVM -- Put "f/2/g"
E (2445432) chip[SVR]: Group removed from nonexistent fabric?
I (2445441) chip[DL]: KVM -- Delete "f/2/k/101"
I (2445452) chip[DL]: KVM -- Put "f/2/g"
I (2445462) chip[DL]: KVM -- Delete "f/2/k/0"
I (2445470) chip[DL]: KVM -- Put "f/2/g"
I (2445477) chip[DL]: KVM -- Put "g/gfl"
I (2445485) chip[DL]: KVM -- Delete "f/2/g"
I (2445498) chip[DL]: KVM -- Delete "f/2/ac/0/0"
I (2445505) chip[DL]: KVM -- Delete "f/2/ac/1"
I (2445514) chip[ZCL]: OpCreds: RemoveFabric successful
```

### Bug prevalence

Whenever I do this

### GitHub hash of the SDK that was being used

v1.2.0.0

### Platform

esp32

### Type

Spec Compliance Issue

zacwbond, 2023-12-01:
> @zacwbond Note that Scenes are provisional in 1.2, and should not be shipped by actual devices.
> 
> Scenes in 1.3 look quite different from scenes in 1.2 (in particular some of the attributes you mention don't even exist).

Thanks for the feedback.  I see the spec says scenes is provisional, but it also says the fan control cluster is provisional, yet our test lab contacts insist that fan control is now certifiable.  So I'm never sure if clusters noted as provisional are actually provisional or not.

(Particularly when revision 5 of the Scenes cluster revision history (1.4.1 of the spec) notes "not provisional.")

lpbeliveau-silabs, 2023-12-07:
As @bzbarsky-apple has pointed out, Scenes were not supported in 1.2.
Specifically, these attributes (SceneCount and Remaining Capacity) were not properly fabric scoped at the time.

Those same attributes are now currently not supported by matter in the upcoming 1.3 spec and have been replaced by their fabric scoped equivalent in the FabricSceneInfo table (as can be seen in the _spec: 1.4.7.1. SceneInfoStruct Type_)

closed 2023-12-07 by lpbeliveau-silabs

bzbarsky-apple, 2023-12-08:
@zacwbond I agree there is a lot of confusion around this, including some of the test labs being confused....

zacwbond, 2023-12-09:
> @zacwbond I agree there is a lot of confusion around this, including some of the test labs being confused....

To anyone else who might be confused in the future about what is and isn't "provisional" in the sense that it can be certified, the answer seems to be in the file `Matter_V_1_2_FallRelease2023_TestPlanVerificationSteps` that comes with the 1.2 spec documents.  There's a sheet titled that lists all available tests, and notes which are and are not verifiable on for the 1.2 release.    I presume future releases will have a similar document.  

And to potentially save someone time, I believe this is the list of unverifiable tests.  Whether a cluster that has some verifiable and some unverifiable tests can be certified, I would guess yes, but I don't know.  

- Bulk Data Exchange Protocol 
   - TC-BDX-1.2, 1.3, 1.6, 2.X
- Bridged Device Basic Information
   - TC-BRBINFO-3.1
- General Commissioning Cluster 
   - TC-CGEN-2.2
- Network Commissioning Cluster
   - TC-CNET-4.13, 4.14
- Device Attestation 
   - TC-DA-1.6
- Device Discovery 
   - TC-DD-1.4, 1.11, 3.2, 3.6, 3.7, 3.8
- Dishwasher Alarm Cluster 
   - TC-DISHALM-3.X
- Diagnostics Logs Cluster 
   - TC-DLOG-X.Y
- Group Communication
   - TC-GRPKEY-5.4
- ICD Management Cluster 
   - TC-ICDM-2.2, 2.3
- Interaction Data Model 
   - TC-IDM-9.1
- Node Operational Credentials Cluster
   - TC-OPCREDS-3.1, 3.4, 3.5
- Power Source Cluster 
   - TC-PS-2.2
- RVC Clean Mode Select Cluster 
   - TC-RVCCLEANM-3.1, 3.3
- RVC Run Mode Select Cluster 
   - TC-RVCRUNM-3.1, 3.3
- Scenes Cluster 
   - TC-S-X.Y
- Secure Channel 
   - TC-SC-1.X, 2.X, 3.1, 3.2, 3.3, 3.4, 3.5, 4.5