project-chip/connectedhomeip #30754: [CI/CD] dependabot CI runs for both push and pull (effectively double the processing)

andy31415, 2023-11-30:
Example here: https://github.com/project-chip/connectedhomeip/pull/30697

We have separate runs on "push on branch" and pr creation, with the intent of "run on master" as well as "run on user PRs".

Dependeabot seems to create branches on the main repo, so the result is that we run 2x the CI. We should only run one

kliao-csa, 2024-03-29:
Chris wanted me to look into this; will update as I dig into what needs to be done here

kliao-csa, 2024-04-02:
Issue: What the issue is: most of our CI runs with the `on: push:` trigger, which will trigger for every commit to the repository. Since contributor changes are performed on forks, `on: push:` does not trigger CI. Instead, CI triggers for changes made in forks through `on: pull_request:` when pull requests from those forks are created. However, dependabot creates branches directly in the main repository instead of being a real user with a fork on its profile. This will trigger `on: push:` whenever it commits and `on: pull_request:` when that branch is pushed to master.

Solution:

1. We can find all CI workflows with `on: push:` and modify it to `on: push: branches-ignore: - 'dependabot/**'`, excluding dependabot branches from triggering CI
2. We can instead use `on: push: branches: - 'master'` to fulfill 

> with the intent of "run on master"

but we will need to confirm that we don't ever want CI to run on branches.

kliao-csa, 2024-04-03:
Vatsal and I split this task up. The two PRs cover the relevant CI that was run twice on the most recent dependabot branch.

To verify, I will create a fake "dependabot/test" branch and commit to it then PR to master. No duplicate CI runs means success. Additionally, I will verify that all CI with `on: push:` triggers either has the master limitation or some other which prevents dependabot runs.

kliao-csa, 2024-04-09:
Some files were missed / newly added that need this fix:

Build example - NXP
Build example - TI CC13XX_26XX
Darwin Tests
Test TV Casting Example
Build example - RW61X

closed 2024-04-17 by kliao-csa