project-chip/connectedhomeip #30756: [Trivial] Enhance debug log details

jvmahon, 2023-11-30:
### Type

Other

### Platform

other

### Platform Version(s)

SDK 1.2.0.1 / All

### Testing

Manually tested with SDK

### Anything else?

Request relates to these lines of code here: 
https://github.com/project-chip/connectedhomeip/blob/ba504e34396545b506cb69adf8aa062cb94815c7/src/lib/dnssd/Resolver_ImplMinimalMdns.cpp#L206

Using the HomeAssistant matter server, I occasionally see floods of debug data (sometimes 50 or so lines) of the form:
`````
2023-11-25 06:34:41 core-matter-server chip.DIS[126] DEBUG SRV record already actively processed.
2023-11-25 06:34:41 core-matter-server chip.DIS[126] DEBUG SRV record already actively processed.
2023-11-25 06:34:41 core-matter-server chip.DIS[126] DEBUG SRV record already actively processed.
`````

I've been informed this is a mDNS problem, and that somewhere on my network a device is creating excess mDNS messages, but I have no way of figuring out which device. To make this easier for users to solve, is it possible to add into the code for this output message an indicator of the device that this relates to (e.g., IP address or anything else).

bzbarsky-apple, 2023-12-04:
@andy31415

jvmahon, 2023-12-04:
Thanks!  To further clarify what I hope to accomplish - I'm getting a flood of these "SRV record already actively processed" messages but don't know whether there is one device somewhere on my network flooding the network, or if this is somehow normal. If these messages would give a bit more info (originator, destination, whatever) allowing one to see where the messages are coming / going, it might help identify if there was a problem device, or if this is just normal operation.

andy31415, 2023-12-05:
We could add some extra logs, however you may also be able to just sniff traffic with wireshark in the mean time.

jvmahon, 2023-12-05:
@andy31415 
Thank you. I think extra information in the log message would be the way to go.

I tried Wireshark, but as a practical matter, it adds a good bit of extra complexity to figure out (certainly much too much complexity for the average consumer). Since  the errors arose using just off-the-shelf products, it would be preferred if the message itself helped more in the diagnostics (at the very least, I'd like to see if its one device spamming the network, or multiple devices in a more "normal" operation).

closed 2024-07-08 by jvmahon