project-chip/connectedhomeip #30781: [IDM-2.2] Wildcard reads failure.

manjunath-grl, 2023-12-04:
### Reproduction steps

Summary:
In IDM-2.2, the 6th step instructs to read any global attribute across all endpoints and clusters. However, when ran using command "ReadById" is executed, the test case fails, accompanied by an error message stating: "KeyError: 'Error: The cluster (0x7) attribute (0xfffd) definition cannot be found. Please update the cluster definition.'

Reproduction steps:
Run TC-IDM_2_2_ script using below command: 
`./scripts/run_in_build_env.sh "./scripts/tests/run_test_suite.py --runner chip_tool_python --chip-tool ./out/linux-x64-chip-tool-no-ble-tsan-clang/chip-tool --target Test_TC_IDM_2_2_ run --iterations 1 --all-clusters-app ./out/linux-x64-all-clusters-no-ble-tsan-clang/chip-all-clusters-app" `

Diff link: https://github.com/project-chip/connectedhomeip/compare/master...manjunath-grl:connectedhomeip:Wildcard_entry_Failures

### Bug prevalence

Everytime when executed

### GitHub hash of the SDK that was being used

2a4a2cada95a8cf0d8447996d47b8cbd77ebd73e

### Platform

other

### Anything else?

Test-plan screenshot:
[image]
Execution logs:
[image]
[image]
[IDM_22_Step6_Fail.txt](https://github.com/project-chip/connectedhomeip/files/13542905/IDM_22_Step6_Fail.txt)

bzbarsky-apple, 2023-12-04:
Cluster 0x7 is "On/Off Switch Configuration".

That said, I have no idea why its global ClusterRevision attribute definition would not be found.   This looks like a likely bug in the test harness.

bzbarsky-apple, 2023-12-04:
Relevant code is at https://github.com/manjunath-grl/connectedhomeip/tree/Wildcard_entry_Failures

bzbarsky-apple, 2023-12-04:
Ah, this is the same issue as https://github.com/project-chip/connectedhomeip/issues/29115

That was closed without fixing the actual issue, so here it is biting us again.  Like that issue says:

> This specific cluster is draft, but this would presumably also fail for manufacturer-specific things.

@cecille

cecille, 2023-12-04:
Other issue was with 1.2 - that was closed because we have a python script that does the same.

But if this is failing on any cluster, this is a real failure. MEI clusters derive from base, and they need the global attributes.

cecille, 2023-12-04:
Oh, sorry, misunderstood - it's not the read that's failing, it's the translation of some sort.

@vivien-apple is probably the best person here to look into this - it is a failure in the yaml runner, not the script.

manjunath-grl, 2023-12-05:
@cecille Tried on ToT able to reproduce it with below command.
` ./scripts/run_in_build_env.sh "./scripts/tests/run_test_suite.py --runner chip_tool_python --chip-tool ./out/linux-x64-chip-tool-no-ble-tsan-clang/chip-tool --target Test_TC_IDM_2_2_ run --iterations 1 --all-clusters-app ./out/linux-x64-all-clusters-no-ble-tsan-clang/chip-all-clusters-app"`

bzbarsky-apple, 2023-12-05:
@cecille You want to pull the code for https://github.com/manjunath-grl/connectedhomeip/tree/Wildcard_entry_Failures

And yes, this is a failure for any unknown cluster; the 2.1 test triggering it was just a symptom.  I should perhaps have been clearer in the other issue...

cecille, 2023-12-05:
OK, the unfortunate part here is that we seem to have lost the main maintainer for the YAML runner. Having me poke around in there is not the best way to get things done since I've literally never touched this code before. None the less, I have prepped a PR that fixes the test in possibly a horrible way. I would like to get feedback from at least @tehampson before unleashing this on the world though as it is more than possible that my confidence outpaces my skill in this area.
https://github.com/project-chip/connectedhomeip/pull/30823

cecille, 2023-12-05:
There's a meta question here, and that is why you're trying to write TC-IDM-2.2 in YAML and whether you think you can succcessfully make it to the end of the test doing all the validation required using only yaml. I would contend that the answer here is no, that this test is better done in python, and we should punt it over to the new CSA staff to automate properly in a way where we can validate the returned values in a reasonable way.

Terence DOES think he can make global attributes work for yaml wildcards. But MEI clusters remain a problem. We could use this override here, but I'm not sure where we have a test case where we're wildcarding on something where we're going to have unknown data coming back.