project-chip/connectedhomeip #30902: [ICD follow-up] If we are getting a check-in, the server thinks we have no subscription to it, right?  Might be worth a followup to think about the logic here.

yunhanw-google, 2023-12-08:
That said, is this the right thing?  If we are getting a check-in, the server thinks we have no subscription to it, right?  Might be worth a followup to think about the logic here.

_Originally posted by @bzbarsky-apple in https://github.com/project-chip/connectedhomeip/pull/30812#discussion_r1420833155_

yunhanw-google, 2024-10-14:
@bzbarsky-apple  is this still a issue? If there is no subscription in server, currently icd server would not send check-in message. cc: @mkardous-silabs

bzbarsky-apple, 2024-10-15:
I don't understand what "If there is no subscription in server, currently icd server would not send check-in message" means.  That's exactly when the server _would_ send a check-in message!

yunhanw-google, 2024-10-15:
sorry, i mistype the sentence, yes,  If there is no subscription in server, currently icd server would send check-in message. Let's make a example, 
assume controller has a active subscription with a lit device, then we suddently reboot that lit device that does not have persistent subscription, a check-in message is sent
a. subscription has not yet timed out, we do nothing here.
b. subscription has timed out,  we didn't get a message from the server on time; it's possible that it no longer has a useful CASE session to us.  Mark defunct all sessions..

It seems current logic looks fine, any missing?

bzbarsky-apple, 2024-10-19:
>  a. subscription has not yet timed out, we do nothing here.

My point is this is wrong.  It hasn't timed out, but the server has just told us "this subscription is gone".  We should be acting on that, I would think, instead of waiting for it to time out and then waiting for the next check-in after _that_ (whenever that is) to re-establish connectivity.

closed 2025-04-03 by yunhanw-google