project-chip/connectedhomeip #30906: [Feature] Integrate tv-app and tv-casting-app into our CI system

yufengwangca, 2023-12-10:
### Feature description

...
we have two Android apps, namely, the tv-app and tv-casting-app, that are not integrated into our CI system. We need to  integrate them into CI to detect any potential issues or breaks. Here is the reference.

full android builds we have here:
https://github.com/project-chip/connectedhomeip/blob/master/.github/workflows/full-android.yaml
smoketest here:
https://github.com/project-chip/connectedhomeip/blob/master/.github/workflows/smoketest-android.yaml
NRF is an example that does path detection to run more things when specific paths are changed:
https://github.com/project-chip/connectedhomeip/blob/master/.github/workflows/examples-nrfconnect.yaml#L51

### Platform

android

kliao-csa, 2024-02-15:
#32012

kliao-csa, 2024-02-16:
Now runs on PR

Next steps:
NRF style path detection
Possibly move code to scripts/examples/tv_app_example.sh to match convention
Remove from smoke tests

kliao-csa, 2024-03-28:
https://github.com/project-chip/connectedhomeip/pull/32774 Reverting changes made so far - I thought we wanted to add smoketest CI builds into standalone non-preliminary CI builds, but it seems we would actually like runtime testing.

kliao-csa, 2024-04-09:
After reading over the additional context provided, I believe this issue is a duplicate of #32787

kliao-csa, 2024-04-09:
As there is already work done on that issue, I believe we can close this duplicate. Second opinions welcome.

kliao-csa, 2024-04-09:
Mark not as complete, but as duplicate

closed 2024-04-09 by kliao-csa