project-chip/connectedhomeip #30908: [BUG] CHIP_ERROR_OUTBOUND_MESSAGE_TOO_BIG when querying attributes

pachuho, 2023-12-11:
### Reproduction steps

1. arm 32-bit environment build
2. Commissioning specific devices (type: thermo-hygrometer, vid:65521, pid: 32773)
3. Attribute Query

```
...
Command failed chip.devicecontroller.ChipClusterException: CHIP cluster error: 195
``` 

### Bug prevalence

About five minutes after the Commissioning

### GitHub hash of the SDK that was being used

2e9c162

### Platform

android

### Anything else?

I'm using an unauthenticated Matter device. However, for other unauthenticated devices (light bulbs), I could see that the attributes are searched normally. I want an answer about what the difference is and why the CHIP_ERROR_OUTBUND_MESSAGE_TOO_BIG error occurs. If possible, please also give me a solution.

CHIPError.h
[image]

bzbarsky-apple, 2023-12-11:
CHIP_ERROR_OUTBOUND_MESSAGE_TOO_BIG can happen for two reasons:

1. Over BLE, if there are chained buffers involved and CompactHead can't fit all the data into a single buffer.
2. Over a UDP transport, if `sendmsg` doesn't send the full message.

The one line from the log above is missing too much context to be able to tell which of those cases it falls into.  Please attach (not paste) complete logs.

pachuho, 2023-12-12:
@bzbarsky-apple 
Thank you for your answer.
It's not what I tested, I requested the log and I'll attach it additionally when I get it. Thank you.

closed 2024-12-19 by pachuho