project-chip/connectedhomeip #30909: [Build] Setting up Python environment

developerchengang, 2023-12-11:
### Build issue(s)

Downloading and installing packages into local source directory:

  Setting up CIPD package manager...done (10.6s)

  Setting up Project actions........skipped (0.1s)

  Setting up Python environment.....[-]

ERROR at //build/config/linux/pkg_config.gni:116:17: Script returned non-zero exit code.

    pkgresult = exec_script(pkg_config_script, args, "value")

                ^----------

Current dir: /home/chengang/Projects/connectedhomeip/.environment/gn_out/

Command: python3 /home/chengang/Projects/connectedhomeip/build/config/linux/pkg-config.py glib-2.0 gio-unix-2.0

Returned 1.

stderr:

Could not run pkg-config.

See //build/chip/linux/BUILD.gn:19:1: whence it was called.

pkg_config("glib") {

^-------------------

See //build/chip/linux/gdbus_library.gni:106:7: which caused the file to be included.

      glib_config,

      ^----------

['gn', 'gen', '/home/chengang/Projects/connectedhomeip/.environment/gn_out', '--args=chip_crypto="boringssl" dir_pigweed="/home/chengang/Projects/connectedhomeip/third_party/pigweed/repo"']

Traceback (most recent call last):

  File "/home/chengang/Projects/connectedhomeip/third_party/pigweed/repo/pw_env_setup/py/pw_env_setup/virtualenv_setup/install.py", line 402, in install_packages

    subprocess.check_call(

  File "/usr/lib/python3.10/subprocess.py", line 369, in check_call

    raise CalledProcessError(retcode, cmd)

subprocess.CalledProcessError: Command '['gn', 'gen', '/home/chengang/Projects/connectedhomeip/.environment/gn_out', '--args=chip_crypto="boringssl" dir_pigweed="/home/chengang/Projects/connectedhomeip/third_party/pigweed/repo"']' returned non-zero exit status 1.

During handling of the above exception, another exception occurred:

Traceback (most recent call last):

  File "/home/chengang/Projects/connectedhomeip/third_party/pigweed/repo/pw_env_setup/py/pw_env_setup/env_setup.py", line 982, in <module>

    sys.exit(main())

  File "/home/chengang/Projects/connectedhomeip/third_party/pigweed/repo/pw_env_setup/py/pw_env_setup/env_setup.py", line 974, in main

    return EnvSetup(**vars(parse())).setup()

  File "/home/chengang/Projects/connectedhomeip/third_party/pigweed/repo/pw_env_setup/py/pw_env_setup/env_setup.py", line 573, in setup

    result = step(spin)

  File "/home/chengang/Projects/connectedhomeip/third_party/pigweed/repo/pw_env_setup/py/pw_env_setup/env_setup.py", line 762, in virtualenv

    if not virtualenv_setup.install(

  File "/home/chengang/Projects/connectedhomeip/third_party/pigweed/repo/pw_env_setup/py/pw_env_setup/virtualenv_setup/install.py", line 443, in install

    install_packages(gn_target)

  File "/home/chengang/Projects/connectedhomeip/third_party/pigweed/repo/pw_env_setup/py/pw_env_setup/virtualenv_setup/install.py", line 410, in install_packages

    raise subprocess.CalledProcessError(

subprocess.CalledProcessError: Command '['gn', 'gen', '/home/chengang/Projects/connectedhomeip/.environment/gn_out', '--args=chip_crypto="boringssl" dir_pigweed="/home/chengang/Projects/connectedhomeip/third_party/pigweed/repo"']' returned non-zero exit status 1.

Installing pip requirements for all...

[notice] A new release of pip is available: 23.2.1 -> 23.3.1

[notice] To update, run: python3 -m pip install --upgrade pip

### Platform

android

developerchengang, 2023-12-11:
need install “sudo apt-get install libglib2.0-dev libgio2.0-cil-dev”

bzbarsky-apple, 2023-12-11:
> need install “sudo apt-get install libglib2.0-dev libgio2.0-cil-dev”

Does https://github.com/project-chip/connectedhomeip/blob/master/docs/guides/BUILDING.md#installing-prerequisites-on-linux need updating?  It mentions libglib2, but not libgio2....

axelnxp, 2024-02-26:
Hi @bzbarsky-apple 

I'm facing the same issue on the top of master, I've installed everything from the BUILDING, and the 2 packages you mentioned.
Still failing with the same issue.
I had to explicitely run `export PKG_CONFIG_PATH=/usr/lib/x86_64-linux-gnu/pkgconfig` to be able to run the bootstrap.
I got the path by running `dpkg -L libglib2.0-dev`

I'm running on Ubuntu 20.04.

andy31415, 2024-10-21:
In ToT you likely require a newer ubuntu: we need python 3.10 or above and ubuntu 20.04 has python 3.8 I believe.

You need to have pkg-config able to find the glib packages on linux, this looks like a O/S configuration issue rather than bootstrap itself.