project-chip/connectedhomeip #30944: [1.3] Persistent Subscription crash in LIT example app

yunhanw-google, 2023-12-12:
### Reproduction steps

Controller side
./out/linux-x64-chip-tool/chip-tool interactive start
pairing onnetwork-long  1 20202021 3840
accesscontrol subscribe-event access-control-entry-changed 5 20 1 0

Server side
 ./out/linux-x64-lit-icd-ipv6only/lit-icd-app

After subscription is established, we restart lit-icd-app
then  lit-icd-app crash wit the below log
Disabling chip_persist_subscriptions can workaround this issue.

### Bug prevalence

consistent

### GitHub hash of the SDK that was being used

918b92b5f22b6f8f4aaa65f07d61369f6b725a9c

### Platform

raspi, core

### Type

Core SDK Crash

### Anything else?

Client log:
[1702343379.315277][75583:75585] CHIP:DMG: Received Command Response Data, Endpoint=0 Cluster=0x0000_003E Command=0x0000_0005
[1702343379.315306][75583:75585] CHIP:CTL: Received certificate signing request from the device
[1702343379.315327][75583:75585] CHIP:CTL: Successfully finished commissioning step 'SendOpCertSigningRequest'
[1702343379.315346][75583:75585] CHIP:CTL: Commissioning stage next step: 'SendOpCertSigningRequest' -> 'ValidateCSR'
[1702343379.315372][75583:75585] CHIP:CTL: Performing next commissioning step 'ValidateCSR'
[1702343379.316205][75583:75585] CHIP:CTL: Successfully finished commissioning step 'ValidateCSR'
[1702343379.316225][75583:75585] CHIP:CTL: Commissioning stage next step: 'ValidateCSR' -> 'GenerateNOCChain'
[1702343379.316252][75583:75585] CHIP:CTL: Performing next commissioning step 'GenerateNOCChain'
[1702343379.316271][75583:75585] CHIP:CTL: Getting certificate chain for the device from the issuer
[1702343379.316490][75583:75585] CHIP:CTL: Verifying Certificate Signing Request
[1702343379.317106][75583:75585] CHIP:CTL: Generating NOC
[1702343379.317304][75583:75585] CHIP:CTL: Providing certificate chain to the commissioner
[1702343379.317326][75583:75585] CHIP:CTL: Received callback from the CA for NOC Chain generation. Status src/controller/ExampleOperationalCredentialsIssuer.cpp:395: Success
[1702343379.317348][75583:75585] CHIP:CTL: Successfully finished commissioning step 'GenerateNOCChain'
[1702343379.317484][75583:75585] CHIP:CTL: Performing next commissioning step 'SendTrustedRootCert'
[1702343379.317503][75583:75585] CHIP:CTL: Sending root certificate to the device
[1702343379.317548][75583:75585] CHIP:DMG: ICR moving to [AddingComm]
[1702343379.317569][75583:75585] CHIP:DMG: ICR moving to [AddedComma]
[1702343379.317671][75583:75585] CHIP:EM: <<< [E:49788i S:61952 M:250313562] (S) Msg TX to 0:FFFFFFFB00000000 [0000] [UDP:[fe80::42:c0ff:fea8:902%eth0]:5540] --- Type 0001:08 (IM:InvokeCommandRequest)
[1702343379.317767][75583:75585] CHIP:DMG: ICR moving to [CommandSen]
[1702343379.317783][75583:75585] CHIP:CTL: Sent root certificate to the device
[1702343379.317944][75583:75585] CHIP:DMG: ICR moving to [AwaitingDe]
[1702343379.318011][75583:75585] CHIP:EM: <<< [E:49787i S:61952 M:250313563 (Ack:185061487)] (S) Msg TX to 0:FFFFFFFB00000000 [0000] [UDP:[fe80::42:c0ff:fea8:902%eth0]:5540] --- Type 0000:10 (SecureChannel:StandaloneAck)
[1702343379.318059][75583:75585] CHIP:EM: Flushed pending ack for MessageCounter:185061487 on exchange 49787i
[1702343379.318790][75583:75585] CHIP:EM: >>> [E:49788i S:61952 M:185061488 (Ack:250313562)] (S) Msg RX from 0:FFFFFFFB00000000 [0000] --- Type 0000:10 (SecureChannel:StandaloneAck)
[1702343379.318808][75583:75585] CHIP:EM: Found matching exchange: 49788i, Delegate: 0x7f6e7c021f18
[1702343379.318829][75583:75585] CHIP:EM: Rxd Ack; Removing MessageCounter:250313562 from Retrans Table on exchange 49788i
[1702343379.319606][75583:75585] CHIP:EM: >>> [E:49788i S:61952 M:185061489 (Ack:250313562)] (S) Msg RX from 0:FFFFFFFB00000000 [0000] --- Type 0001:09 (IM:InvokeCommandResponse)
[1702343379.319624][75583:75585] CHIP:EM: Found matching exchange: 49788i, Delegate: 0x7f6e7c021f18
[1702343379.319640][75583:75585] CHIP:EM: CHIP MessageCounter:250313562 not in RetransTable on exchange 49788i
[1702343379.319664][75583:75585] CHIP:DMG: ICR moving to [ResponseRe]
[1702343379.319691][75583:75585] CHIP:DMG: InvokeResponseMessage =
[1702343379.319704][75583:75585] CHIP:DMG: {
[1702343379.319720][75583:75585] CHIP:DMG: 	suppressResponse = false, 
[1702343379.319735][75583:75585] CHIP:DMG: 	InvokeResponseIBs =
[1702343379.319757][75583:75585] CHIP:DMG: 	[
[1702343379.319770][75583:75585] CHIP:DMG: 		InvokeResponseIB =
[... 598 lines cut ...]
#17 0x000055c3cb60d071 in chip::OperationalSessionSetup::UpdateDeviceData(chip::Transport::PeerAddress const&, chip::ReliableMessageProtocolConfig const&)
    (this=0x55c3cc75fc00, addr=..., config=...) at ../../examples/lit-icd-app/linux/third_party/connectedhomeip/src/app/OperationalSessionSetup.cpp:232
#18 0x000055c3cb60e32b in chip::OperationalSessionSetup::OnNodeAddressResolved(chip::PeerId const&, chip::AddressResolve::ResolveResult const&) (this=0x55c3cc75fc00, peerId=..., result=...)
    at ../../examples/lit-icd-app/linux/third_party/connectedhomeip/src/app/OperationalSessionSetup.cpp:572
#19 0x000055c3cb6cc4f6 in chip::AddressResolve::Impl::Resolver::HandleAction(chip::IntrusiveList<chip::AddressResolve::Impl::NodeLookupHandle, (chip::IntrusiveMode)0, chip::IntrusiveListBase--Type <RET> for more, q to quit, c to continue without paging--
Hook<chip::AddressResolve::Impl::NodeLookupHandle, (chip::IntrusiveMode)0> >::Iterator&) (this=0x55c3cb7e1ce0 <chip::AddressResolve::Resolver::Instance()::gResolver>, current=...)
    at ../../examples/lit-icd-app/linux/third_party/connectedhomeip/src/lib/address_resolve/AddressResolve_DefaultImpl.cpp:342
#20 0x000055c3cb6cc5cc in chip::AddressResolve::Impl::Resolver::HandleTimer() (this=0x55c3cb7e1ce0 <chip::AddressResolve::Resolver::Instance()::gResolver>)
    at ../../examples/lit-icd-app/linux/third_party/connectedhomeip/src/lib/address_resolve/AddressResolve_DefaultImpl.cpp:358
#21 0x000055c3cb6cb154 in chip::AddressResolve::Impl::Resolver::OnResolveTimer(chip::System::Layer*, void*)
    (layer=0x55c3cb7dd4a0 <chip::DeviceLayer::SystemLayerImpl()::gSystemLayerImpl>, context=0x55c3cb7e1ce0 <chip::AddressResolve::Resolver::Instance()::gResolver>)
    at ../../examples/lit-icd-app/linux/third_party/connectedhomeip/src/lib/address_resolve/AddressResolve_DefaultImpl.h:185
#22 0x000055c3cb6492cf in chip::System::TimerData::Callback::Invoke() const (this=0x7ffc48ccd590) at ../../examples/lit-icd-app/linux/third_party/connectedhomeip/src/system/SystemTimer.h:61
#23 0x000055c3cb64adb6 in chip::System::TimerPool<chip::System::TimerList::Node>::Invoke(chip::System::TimerList::Node*)
    (this=0x55c3cb7ddaa8 <chip::DeviceLayer::SystemLayerImpl()::gSystemLayerImpl+1544>, timer=0x55c3cc763d30)
    at ../../examples/lit-icd-app/linux/third_party/connectedhomeip/src/system/SystemTimer.h:237
#24 0x000055c3cb64a964 in chip::System::LayerImplSelect::HandleEvents() (this=0x55c3cb7dd4a0 <chip::DeviceLayer::SystemLayerImpl()::gSystemLayerImpl>)
    at ../../examples/lit-icd-app/linux/third_party/connectedhomeip/src/system/SystemLayerImplSelect.cpp:678
#25 0x000055c3cb689eb4 in chip::DeviceLayer::Internal::GenericPlatformManagerImpl_POSIX<chip::DeviceLayer::PlatformManagerImpl>::_RunEventLoop()
    (this=0x55c3cb7e1350 <chip::DeviceLayer::PlatformManagerImpl::sInstance+16>)
    at ../../examples/lit-icd-app/linux/third_party/connectedhomeip/src/include/platform/internal/GenericPlatformManagerImpl_POSIX.ipp:225
#26 0x000055c3cb41c6d8 in chip::DeviceLayer::PlatformManager::RunEventLoop() (this=0x55c3cb7e1340 <chip::DeviceLayer::PlatformManagerImpl::sInstance>)
    at ../../examples/lit-icd-app/linux/third_party/connectedhomeip/src/include/platform/PlatformManager.h:403
#27 0x000055c3cb41f1bf in ChipLinuxAppMainLoop(AppMainLoopImplementation*) (impl=0x0) at ../../examples/lit-icd-app/linux/third_party/connectedhomeip/examples/platform/linux/AppMain.cpp:609
#28 0x000055c3cb3c21eb in main(int, char**) (argc=1, argv=0x7ffc48ccdaa8) at ../../examples/lit-icd-app/linux/main.cpp:32

bzbarsky-apple, 2023-12-12:
Pinging @jtung-apple 

That said, @yunhanw-google this sounds like "some code is asking for subjects on subscriptions that don't have a CASE session", which is a problem in that code, no?  CASE sessions can be evicted for existing subscriptions completely independent of persistent subscriptions, right?  Or does that case immediately tear down the subscription?

Note that https://github.com/project-chip/connectedhomeip/pull/30491 may well fix this for the persistent subscription case, but might not fix the real underlying issue.

mkardous-silabs, 2024-01-22:
Could this be related to https://github.com/project-chip/connectedhomeip/issues/30518?