project-chip/connectedhomeip #30973: [1.1] Spake2p::ComputeRoundTwo function can not successed process

wulaidehouzi, 2023-12-13:
### Reproduction steps

When I tried to debug the lighting routine of matter on Ubuntu, CHIPTool had the following error during the PASE process: CHIP:SC: Failed during PASE session setup: src/crypto/CHIPCryptoPALOpenSSL.cpp:1491: CHIP Error 0x000000AC:  Internal error
The error was returned by PointLoad when Ubuntu entered Spake2p::ComputeRoundTwo. The PointLoad function called Openssl library through EC_POINT_oct2point. EC_POINT_oct2point returned an error, is there a problem with my data?

### Bug prevalence

Whatever i do this

### GitHub hash of the SDK that was being used

17a8207a16df2965e77c6bbe869c6dcc9bfb5f80

### Platform

other

### Platform Version(s)

1.1

### Type

Core SDK Crash

Alami-Amine, 2025-05-16:
This is strange actually, do you still reproduce this?

Alami-Amine, 2025-05-16:
what Passcode did you use?