project-chip/connectedhomeip #30981: [BUG] Activate script is failing due to python dependencies to prompt_toolkit

GabrielCouturier, 2023-12-13:
### Reproduction steps

Please see the comments 

### Bug prevalence

always 

### GitHub hash of the SDK that was being used

a9ad2c49198a5f45c604d55f31ee987e678165ab

### Platform

k32w

### Anything else?

Hello,

When running the ./script/activate.sh after cloning the repos we have the following issue : 
see build example of this job https://github.com/project-chip/connectedhomeip/actions/runs/7196189624/job/19600480143 

Checking the environment:

Traceback (most recent call last):
  File "/__w/connectedhomeip/connectedhomeip/.environment/pigweed-venv/lib/python3.9/site-packages/pw_cli/plugins.py", line 96, in from_name
    module = importlib.import_module(module_name)
  File "/usr/lib/python3.9/importlib/__init__.py", line 127, in import_module
    return _bootstrap._gcd_import(name[level:], package, level)
  File "<frozen importlib._bootstrap>", line 1030, in _gcd_import
  File "<frozen importlib._bootstrap>", line 1007, in _find_and_load
  File "<frozen importlib._bootstrap>", line 986, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 680, in _load_unlocked
  File "<frozen importlib._bootstrap_external>", line 8[55](https://github.com/project-chip/connectedhomeip/actions/runs/7196189624/job/19600480143#step:6:56), in exec_module
  File "<frozen importlib._bootstrap>", line 228, in _call_with_frames_removed
  File "/__w/connectedhomeip/connectedhomeip/.environment/pigweed-venv/lib/python3.9/site-packages/pw_watch/watch.py", line 88, in <module>
    import pw_console.python_logging
  File "/__w/connectedhomeip/connectedhomeip/.environment/pigweed-venv/lib/python3.9/site-packages/pw_console/__init__.py", line 16, in <module>
    from pw_console.embed import PwConsoleEmbed
  File "/__w/connectedhomeip/connectedhomeip/.environment/pigweed-venv/lib/python3.9/site-packages/pw_console/embed.py", line 23, in <module>
    from pw_console.console_app import ConsoleApp
  File "/__w/connectedhomeip/connectedhomeip/.environment/pigweed-venv/lib/python3.9/site-packages/pw_console/console_app.py", line [59](https://github.com/project-chip/connectedhomeip/actions/runs/7196189624/job/19600480143#step:6:60), in <module>
    from ptpython.layout import CompletionVisualisation  # type: ignore
  File "/__w/connectedhomeip/connectedhomeip/.environment/pigweed-venv/lib/python3.9/site-packages/ptpython/__init__.py", line 3, in <module>
    from .repl import embed
  File "/__w/connectedhomeip/connectedhomeip/.environment/pigweed-venv/lib/python3.9/site-packages/ptpython/repl.py", line 23, in <module>
    from prompt_toolkit.formatted_text import OneStyleAndTextTuple
ImportError: cannot import name 'OneStyleAndTextTuple' from 'prompt_toolkit.formatted_text' (/__w/connectedhomeip/connectedhomeip/.environment/pigweed-venv/lib/python3.9/site-packages/prompt_toolkit/formatted_text/__init__.py)

It seems to be linked to :
https://github.com/project-chip/connectedhomeip/blob/master/examples/common/pigweed/rpc_console/py/setup.cfg
Where the prompt_toolkit version is not fixed.

The prompt_toolkit has been updated this morning : 
https://pypi.org/project/prompt-toolkit/#history

caipiblack, 2023-12-19:
Is it possible to fix it for the tag v1.2.0.1 ? Or create a new tag ?

agners, 2023-12-20:
So it seems something changed in Pigweed which made this suddenly fail? Is it maybe possible to pin to a stable Pigweed version? :thinking:

andy31415, 2024-01-11:
> So it seems something changed in Pigweed which made this suddenly fail? Is it maybe possible to pin to a stable Pigweed version? 🤔

We are pinning pigweed in submodules - I am not sure about the actual root cause.

PR #30987 (2023-12-13): Bump prompt-toolkit requirement to fix ptpython version update
Fixes #30981