project-chip/connectedhomeip #30989: [BUG] Vendor-specific fields not fully tested in SDK

mlepage-google, 2023-12-13:
### Reproduction steps

Test cluster (0xFFF1FC05) (src/app/zap-templates/zcl/data-model/chip/test-cluster.xml) does not seem to have cases where we have vendor-specific fields in attribute structs, event structs, or command argument structs. These are needed to test MEI functionality in the SDK.

E.g. NestedStruct tests that SimpleStruct can be nested within it, but doesn't test that:

1) Another vendor (e.g. 0xFFF2) can add a field to NestedStruct
2) Another vendor (e.g. 0xFFF2 or 0xFFF3) can add a field to SimpleStruct, which is nested within NestedStruct

### Bug prevalence

Always

### GitHub hash of the SDK that was being used

Latest (and presumably all)

### Platform

core

bzbarsky-apple, 2023-12-13:
None of that is supported right now, basically.  Which is why the spec has:

>  Support for encoding Field IDs with an MC source is provisional. 

Because the SDK does not actually let you do that.

tcarmelveilleux, 2023-12-14:
> Because the SDK does not actually let you do that.

With direct TLV access:

```
void ReadClient::Callback::OnAttributeData(const ConcreteDataAttributePath & aPath, TLV::TLVReader * apData, const StatusIB & aStatus) {}
```

and also with AttributeAccessInterface, it is possible for a client to see these fields as per spec.

bzbarsky-apple, 2023-12-16:
That's true.  You just can't do it via the codegen helpers.