project-chip/connectedhomeip #30995: [BUG] bootstrap.sh run SHA 0ea15eb9d6a8ac35cfc8148e654013370d1e6e8b shows Error: Failed to import module "pw_watch.watch"

vinay-apple, 2023-12-13:
### Reproduction steps

1. clone the chip repo 
2. point to the SHA 0ea15eb9d6a8ac35cfc8148e654013370d1e6e8b
3. run the source scripts/bootstrap.sh to setup the environment and see that it fails with below error (detailed logs attached as text as well) 

Activating environment (setting environment variables):

  Setting environment variables for CIPD package manager...done
  Setting environment variables for Project actions........skipped
  Setting environment variables for Python environment.....done
  Setting environment variables for pw packages............skipped
  Setting environment variables for Host tools.............done

Checking the environment:

Traceback (most recent call last):
  File "/Users/vinayganji/CHIP-REPO/connectedhomeip/.environment/pigweed-venv/lib/python3.11/site-packages/pw_cli/plugins.py", line 96, in from_name
    module = importlib.import_module(module_name)
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/opt/homebrew/Cellar/python@3.11/3.11.6_1/Frameworks/Python.framework/Versions/3.11/lib/python3.11/importlib/__init__.py", line 126, in import_module
    return _bootstrap._gcd_import(name[level:], package, level)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "<frozen importlib._bootstrap>", line 1204, in _gcd_import
  File "<frozen importlib._bootstrap>", line 1176, in _find_and_load
  File "<frozen importlib._bootstrap>", line 1147, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 690, in _load_unlocked
  File "<frozen importlib._bootstrap_external>", line 940, in exec_module
  File "<frozen importlib._bootstrap>", line 241, in _call_with_frames_removed
  File "/Users/vinayganji/CHIP-REPO/connectedhomeip/.environment/pigweed-venv/lib/python3.11/site-packages/pw_watch/watch.py", line 88, in <module>
    import pw_console.python_logging
  File "/Users/vinayganji/CHIP-REPO/connectedhomeip/.environment/pigweed-venv/lib/python3.11/site-packages/pw_console/__init__.py", line 16, in <module>
    from pw_console.embed import PwConsoleEmbed
  File "/Users/vinayganji/CHIP-REPO/connectedhomeip/.environment/pigweed-venv/lib/python3.11/site-packages/pw_console/embed.py", line 23, in <module>
    from pw_console.console_app import ConsoleApp
  File "/Users/vinayganji/CHIP-REPO/connectedhomeip/.environment/pigweed-venv/lib/python3.11/site-packages/pw_console/console_app.py", line 59, in <module>
    from ptpython.layout import CompletionVisualisation  # type: ignore
    ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/Users/vinayganji/CHIP-REPO/connectedhomeip/.environment/pigweed-venv/lib/python3.11/site-packages/ptpython/__init__.py", line 3, in <module>
    from .repl import embed
  File "/Users/vinayganji/CHIP-REPO/connectedhomeip/.environment/pigweed-venv/lib/python3.11/site-packages/ptpython/repl.py", line 23, in <module>
    from prompt_toolkit.formatted_text import OneStyleAndTextTuple
ImportError: cannot import name 'OneStyleAndTextTuple' from 'prompt_toolkit.formatted_text' (/Users/vinayganji/CHIP-REPO/connectedhomeip/.environment/pigweed-venv/lib/python3.11/site-packages/prompt_toolkit/formatted_text/__init__.py)

The above exception was the direct cause of the following exception:

Traceback (most recent call last):
  File "/Users/vinayganji/CHIP-REPO/connectedhomeip/.environment/pigweed-venv/bin/pw", line 8, in <module>
    sys.exit(main())
             ^^^^^^
  File "/Users/vinayganji/CHIP-REPO/connectedhomeip/.environment/pigweed-venv/lib/python3.11/site-packages/pw_cli/__main__.py", line 43, in main
    pw_command_plugins.register()
  File "/Users/vinayganji/CHIP-REPO/connectedhomeip/.environment/pigweed-venv/lib/python3.11/site-packages/pw_cli/pw_command_plugins.py", line 65, in register
    _register_builtin_plugins(plugin_registry)
  File "/Users/vinayganji/CHIP-REPO/connectedhomeip/.environment/pigweed-venv/lib/python3.11/site-packages/pw_cli/pw_command_plugins.py", line 43, in _register_builtin_plugins
    registry.register_by_name('watch', 'pw_watch.watch', 'main')
  File "/Users/vinayganji/CHIP-REPO/connectedhomeip/.environment/pigweed-venv/lib/python3.11/site-packages/pw_cli/plugins.py", line 303, in register_by_name
    Plugin.from_name(name, module_name, member_name, source)
  File "/Users/vinayganji/CHIP-REPO/connectedhomeip/.environment/pigweed-venv/lib/python3.11/site-packages/pw_cli/plugins.py", line 104, in from_name
    raise Error(f'Failed to import module "{module_name}"') from err
pw_cli.plugins.Error: Failed to import module "pw_watch.watch" (ImportError: cannot import name 'OneStyleAndTextTuple' from 'prompt_toolkit.formatted_text' (/Users/vinayganji/CHIP-REPO/connectedhomeip/.environment/pigweed-venv/lib/python3.11/site-packages/prompt_toolkit/formatted_text/__init__.py))
Error during bootstrap--see messages above.

 
[bootstrap-issue-0ea15eb9d6a8ac35cfc8148e654013370d1e6e8b.txt](https://github.com/project-chip/connectedhomeip/files/13665818/bootstrap-issue-0ea15eb9d6a8ac35cfc8148e654013370d1e6e8b.txt)

### Bug prevalence

always seeing this issue with the SHA mentioned

### GitHub hash of the SDK that was being used

0ea15eb9d6a8ac35cfc8148e654013370d1e6e8b

### Platform

darwin

ybarigou, 2023-12-15:
Same here, it's recent. Been banging my head to get it fixed/workaround no success. Anyone else seeing this ?

caipiblack, 2023-12-19:
Maybe related to this:  https://github.com/project-chip/connectedhomeip/issues/30981

caipiblack, 2023-12-21:
And this commit resolve the problem:

https://github.com/project-chip/connectedhomeip/pull/30987/files

You just have to apply it, do the bootstrap and activate.sh after changing this file.

andy31415, 2024-01-11:
Duplicate of https://github.com/project-chip/connectedhomeip/issues/30981 - please apply https://github.com/project-chip/connectedhomeip/pull/30987

closed 2024-01-11 by andy31415