project-chip/connectedhomeip #31012: Handle all derived cluster enums together.

hicklin, 2023-12-14:
In derived clusters, the same emun is represented in a different namespaces for the different clusters. This means that functions that take these enums have to use the underlying `uint8_t` type. This can make the code difficult to read due to all the `to_underlying` calls.

Look into the possibility of standardising how these enums are handled such that SDK users do not need to call `to_underlying` when passing these enums.

It would be ideal if this solution would also allow SDK users to hold these enums in the same variable.

bzbarsky-apple, 2023-12-15:
@hasty has something that might fix this.

hasty, 2023-12-15:
Yes, I submitted a PR on zap [here](https://github.com/project-chip/zap/pull/1240), though someone with write access will need to merge it. I'll have another one for this repo once that one bumps the zap release version.