project-chip/connectedhomeip #31020: [Documentation] Usage of StatusCode INVALID_DATA_TYPE (0x8D)

Apollon77, 2023-12-14:
### Documentation issues

The specificaton defines the statuscode 0x8D INVALID_DATA_TYPE with

> The data type indicated is unde­ fined or invalid for the indi­ cated data field. Command or action not carried out.

This statuscode is not mentioned anywhere else in the whole specification (at least not via PDF search, also for parts). So it is officially unclear in which cases to use it and what to expect.

But: In chip repo code this seems to be mapped to EMBER_ZCL_STATUS_INVALID_DATA_TYPE and is used when datatypes are not matching to expected data or such (at least how I understood the code), but seems to be used for attributes and also commands.

With this I expect that Specification misses information on how and when to use this status code and I think this should be added. Additionally I would also be happy to get the info for implementation in matter.js.

### Platform

other

Apollon77, 2023-12-14:
@bzbarsky-apple Should be simple to answer for you I assume :-)

bzbarsky-apple, 2023-12-15:
> and is used when datatypes are not matching to expected data

It's basically used for internal programming errors (e.g. asking to place a 4-byte into into a smaller-than-4-byte buffer).

In practice, given a non-buggy server there should be no way for a client to trigger this status, since as you note the spec never says to return it.

closed 2023-12-15 by bzbarsky-apple

Apollon77, 2023-12-15:
Ok, thank you. This clarified it.