project-chip/connectedhomeip #31027: [BUG] `new` memory allocation raises exception in case of no memory

gautam-dev-maker, 2023-12-15:
### Reproduction steps

# Steps for bug reproduction: 

* Increase number of wifi AP scan list in the environment
* Commission with ecosystem that requires the wifi AP scanlist.

[crash_log.txt](https://github.com/project-chip/connectedhomeip/files/13682721/crash_log.txt)

# Issue

* When utilizing the `new` keyword for memory allocation on the ESP32-C2, encountering a lack of available memory triggers an exception. If this exception remains unhandled, it leads to a system abort, resulting in a device crash.

* During the allocation of memory for storing Wi-Fi Access Points (APs) on the ESP32-C2 ([here](https://github.com/espressif/connectedhomeip/blob/9ec9ce24a7fa33de711dfc7bd26a969000a9b672/src/platform/ESP32/NetworkCommissioningDriver.cpp#L348)), the firmware crashes when memory is insufficient. This occurrence is frequent due to the high memory consumption of the application.

* As per the [crash_log](https://github.com/project-chip/connectedhomeip/files/13682721/crash_log.txt), the largest free block is `5632`, but the required memory is `6552 (78 * 84)`.

* Although error handling mechanisms are in place to verify memory allocation, the code path does not reach the corresponding block as the exception causes an abrupt termination.

# Possible Fix:
Use [std::throw](https://en.cppreference.com/w/cpp/memory/new/nothrow) to overload the error throwing class, it will return null if no memory can be allocated.

### Bug prevalence

Everytime when the memory is not available.

### GitHub hash of the SDK that was being used

9ec9ce24a7fa33de711dfc7bd26a969000a9b672

### Platform

esp32, core

dhrishi, 2023-12-15:
@bzbarsky-apple I assume, there can be some other places in the common code too where this can happen

``` 
    std::unique_ptr<wifi_ap_record_t[]> ap_buffer_ptr(new wifi_ap_record_t[ap_number]);
    if (ap_buffer_ptr == NULL)
```
Have we encountered this before? Is using `std::throw` the solution in the SDK?

bzbarsky-apple, 2023-12-15:
@dhrishi Common code that runs on constrained devices should never use `new`, precisely because of things like this.  It uses `Platform::MemoryAlloc`.

But yes, if you plan to use `new`, then you should `std::nothrow` as needed.

dhrishi, 2023-12-15:
Ok. Thanks for the information. We will make the required changes.

tcarmelveilleux, 2024-01-10:
Also, storing a ton of Wi-Fi scan records is problematic. Suggest keeping memory/slots to ~2k since by the end of it, only a small number of entries can fit in the ScanNetworks