project-chip/connectedhomeip #31041: [Platform]  CC2652R7 Contact Sensor. I can't get it to compile.

Ryekia88, 2023-12-15:
### Reproduction steps

I have been attempting to make a contact sensor. I know that there is no example for the CC2652 boards or any TI boards. I attempted to port the contact sensor from the sample that is there. But, I can not get it to work. I tried to look at the all-cluster example and the lock sample, but both require a cluster server. I looked there and there is not one there for contact-sensor nor boolean. I also noticed that in the common for contact sensor it does not have a make file or the end points that the lock-app has. Do I need those? Sorry, I am new at this. I have previously done programming in other languages, but I have never done anything with matter nor c++. I am trying to make a sensor for a device that I make. The device would send the signal to the matter device to notify the users phone. I was able to add an input to the lock-app, but it does not do what I need it to do. 

### Platform

cc13x2_cc26x2

### Type

Platform validated

bzbarsky-apple, 2023-12-15:
@Ryekia88 For a contact sensor there really isn't any implementation code other than the generic attribute read/write bits, right?

What exact compile failures are you seeing?  Please attach (not paste) logs.

Ryekia88, 2023-12-15:
[log.docx](https://github.com/project-chip/connectedhomeip/files/13688644/log.docx)

Thanks for getting back with me. I didn't think that it needed a server. I tried to change the example that was provided to my board (the example is nxp my board is Texas Instruments).

bzbarsky-apple, 2023-12-15:
OK, so the compile errors look like:
```
./../third_party/connectedhomeip/examples/contact-sensor-app/cc13x2x7_26x2x7/main/ContactSensorManager.cpp:54:43: error: 'kContact' is not a member of 'AppEvent'
   54 |     event.Type                = AppEvent::kContact;
      |                                           ^~~~~~~~
../../third_party/connectedhomeip/examples/contact-sensor-app/cc13x2x7_26x2x7/main/ContactSensorManager.cpp:55:11: error: 'struct AppEvent' has no member named 'ContactEvent'
   55 |     event.ContactEvent.Action = static_cast<uint8_t>(aAction);
      |           ^~~~~~~~~~~~
```

it looks like that code is assuming some ContactEvent thing that is not defined anywhere?

And separately:

```
../../third_party/connectedhomeip/examples/contact-sensor-app/cc13x2x7_26x2x7/main/main.cpp:33:10: fatal error: ti/drivers/UART.h: No such file or directory
   33 | #include <ti/drivers/UART.h>
      |          ^~~~~~~~~~~~~~~~~~~
```

which seems like an issue with your TI sdk bits?

Ryekia88, 2023-12-15:
I don't know why it says its not there. It says that it cant find AppConfig.h and  <platform/cc13x2_26x2/OTAImageProcessorImpl.h>. I based it off of the other example made for the other chip, but I am thinking about starting from scratch. I don't know what else to do.

Ryekia88, 2023-12-20:
I totally started all over and wrote my own files based on the Contact example and the lock examples to create a TI version. I am getting an error at build, it says that their is a Value collision. That is what I am trying to solve right now.

Ryekia88, 2023-12-28:
[[](url)](https://github.com/Ryekia88/Contact_sensor)

That is what I have so far. I can get it to start building, but not finish.

Ryekia88, 2023-12-28:
The good news is i got it to finally start to build, but it has errors that I still have to work through. The first is #include <ti/drivers/UART.h> which it says has no file nor directory.

ReneJosefsen, 2024-01-03:
There were some changes to the UART drivers in a SDK update, where it now only uses/supports UART2 driver. This does require some minor changes to the files.

Ryekia88, 2024-01-24:
So I kept trying with the contact app I made and couldn't get anywhere. So I tried yours and it wont compile. I get a compile error of "if (custom_factory_data)" and the other is "public_deps += [_platform_target]". I have tried working through it, but it doesn't compile. What could I be missing?

Ryekia88, 2024-01-25:
I am sorry for bothering you, I haven't coded in this language before. Most of my time has been in Java, Python and FLutter/dart. So, I am trying to learn how to do this. But, I am sorry for bothering you, it is greatly appreciated.

ReneJosefsen, 2024-01-25:
Not a problem, I kind of had to get these examples updated anyways I just haven't gotten around to do it. Give me a day or two and I will give it a look 😄

Ryekia88, 2024-01-25:
I found out how to get the out/debug and the targets created. In the args.ni file at the very bottom it was missing the line "custom_factory_data = true".

Ryekia88, 2024-01-25:
Good news, I got it to compile and its working on the board. It just says its offline. But, I am just glad to see that something happened. Thank you again.

Ryekia88, 2024-01-26:
So, a quick update. I got it to work and compile and flash to the device. I can get it to connect to my nest, but with a few problems like the qr code not working and that device doesn't stay connected to matter. I just wanted to give you an update on the progress.

Ryekia88, 2024-01-26:
Alright. I solved the connection problem, the qr still is weird and it takes a long time for the sensor to say that it is occupied. Is that normal??

ReneJosefsen, 2024-01-30:
The attribute should change as soon as the button is pressed. We are talking about the occupancy sensor example here right?

Ryekia88, 2024-02-12:
Yea, I got it to be accepted but when I push the button, it doesn't send a signal to alert my nest hub or my phone. I also can't get it set up with the QR code still.

Ryekia88, 2024-02-12:
And now it wont even connect. This is ridiculous. I follow what TI says in their examples and it doesn't work. The haven't updated their documents in forever and you cant get any help from them on how their own product works. They send you to their stupid forums where no one helps you out. I am about to switch to a different company or tell my company that we aren't going to invest in Matter. Nothing works. They provide no information on anything, i have to search through document after document to find out information and it still doesn't work. This is the worst chip that I have ever programmed period.

ReneJosefsen, 2024-02-14:
Do you have any logs where it fails to connect? It is a bit easier to help out if I have a log to see where the failure is happening.

Ryekia88, 2024-02-19:
IM RH moving to [AwaitingDestruction]                                           
All ReadHandler-s are clean, clear GlobalDirtySet                               
>>> [E:59483r S:54039 M:166861624] (S) Msg RX from 0:FFFFFFFB00000000 [0000] --)
Handling via exchange: 59483r, Delegate: 0x20009afc                             
Received Read request                                                           
IM RH moving to [CanStartReporting]                                             
Building Reports for ReadHandler with LastReportGeneration = 0x00000000000000000
<RE:Run> Cluster 28, Attribute 4 is dirty                                       
Reading attribute: Cluster=0x0000_0028 Endpoint=0 AttributeId=0x0000_0004 (expa)
<RE> Sending report (payload has 37 bytes)...                                   
<<< [E:59483r S:54039 M:20871314] (S) Msg TX to 0:FFFFFFFB00000000 [0000] [BLE])
<RE> OnReportConfirm: NumReports = 0                                            
<RE> ReportsInFlight = 0 with readHandler 0, RE has no more messages            
IM RH moving to [AwaitingDestruction]                                           
All ReadHandler-s are clean, clear GlobalDirtySet                               
>>> [E:59484r S:54039 M:166861625] (S) Msg RX from 0:FFFFFFFB00000000 [0000] --)
Handling via exchange: 59484r, Delegate: 0x20009afc                             
Received Read request                                                           
IM RH moving to [CanStartReporting]                                             
Building Reports for ReadHandler with LastReportGeneration = 0x00000000000000000
<RE:Run> Cluster 3e, Attribute 2 is dirty                                       
Reading attribute: Cluster=0x0000_003E Endpoint=0 AttributeId=0x0000_0002 (expa)
<RE> Sending report (payload has 36 bytes)...                                   
<<< [E:59484r S:54039 M:20871315] (S) Msg TX to 0:FFFFFFFB00000000 [0000] [BLE])
<RE> OnReportConfirm: NumReports = 0                                            
<RE> ReportsInFlight = 0 with readHandler 0, RE has no more messages            
IM RH moving to [AwaitingDestruction]                                           
All ReadHandler-s are clean, clear GlobalDirtySet                               
>>> [E:59485r S:54039 M:166861626] (S) Msg RX from 0:FFFFFFFB00000000 [0000] --)
Handling via exchange: 59485r, Delegate: 0x20009afc                             
Received Read request                                                           
IM RH moving to [CanStartReporting]                                             
Building Reports for ReadHandler with LastReportGeneration = 0x00000000000000000
<RE:Run> Cluster 3e, Attribute 3 is dirty                                       
Reading attribute: Cluster=0x0000_003E Endpoint=0 AttributeId=0x0000_0003 (expa)
<RE> Sending report (payload has 36 bytes)...                                   
<<< [E:59485r S:54039 M:20871316] (S) Msg TX to 0:FFFFFFFB00000000 [0000] [BLE])
<RE> OnReportConfirm: NumReports = 0                                            
<RE> ReportsInFlight = 0 with readHandler 0, RE has no more messages            
IM RH moving to [AwaitingDestruction]                                           
All ReadHandler-s are clean, clear GlobalDirtySet                               
>>> [E:59486r S:54039 M:166861627] (S) Msg RX from 0:FFFFFFFB00000000 [0000] --)
Handling via exchange: 59486r, Delegate: 0x20009afc                             
Received command for Endpoint=0 Cluster=0x0000_003E Command=0x0000_0002         
OpCreds: Certificate Chain request received for DAC                             
Command handler moving to [NewRespons]                                          
Command handler moving to [ Preparing]                                          
Command handler moving to [AddingComm]                                          
Command handler moving to [AddedComma]                                          
Decreasing reference count for CommandHandler, remaining 0                      
<<< [E:59486r S:54039 M:20871317] (S) Msg TX to 0:FFFFFFFB00000000 [0000] [BLE])
Command response sender moving to [AllInvokeR]                                  
Command handler moving to [AwaitingDe]                                          
>>> [E:59487r S:54039 M:166861628] (S) Msg RX from 0:FFFFFFFB00000000 [0000] --)
Handling via exchange: 59487r, Delegate: 0x20009afc                             
Received Read request                                                           
IM RH moving to [CanStartReporting]                                             
Building Reports for ReadHandler with LastReportGeneration = 0x00000000000000000
<RE:Run> Cluster 1d, Attribute 1 is dirty                                       
Reading attribute: Cluster=0x0000_001D Endpoint=0 AttributeId=0x0000_0001 (expa)
<RE> Sending report (payload has 60 bytes)...                                   
<<< [E:59487r S:54039 M:20871318] (S) Msg TX to 0:FFFFFFFB00000000 [0000] [BLE])
<RE> OnReportConfirm: NumReports = 0                                            
<RE> ReportsInFlight = 0 with readHandler 0, RE has no more messages            
IM RH moving to [AwaitingDestruction]                                           
All ReadHandler-s are clean, clear GlobalDirtySet                               
>>> [E:59488r S:54039 M:166861629] (S) Msg RX from 0:FFFFFFFB00000000 [0000] --)
[... cut ...]
Pending Last Known Good Time: 2023-10-14T01:16:48                               
Previous Last Known Good Time: 2023-10-14T01:16:48                              
Reverted Last Known Good Time to previous value                                 
Endpoint 0, Cluster 0x0000_0030 update version to f180f4a9                      
Failsafe timeout, tell platform driver to revert network credentials.           
Long dispatch time: 144 ms, for event type 32783                                
Fail-safe cleanly disarmed

ReneJosefsen, 2024-02-20:
The main thing I notice here is the:
`Failed NOC chain validation: 50`

That indicates CHIP_ERROR_UNSUPPORTED_CERT_FORMAT 🤔 I will need to setup a system and test with one of the CC2652R7 LP I have before I can verify any of this. Can you share some details about your setup, what are you using as Thread Border Router and what are you using as controller, chip-tool?

ReneJosefsen, 2024-02-20:
So I tried to build the latest master for some of my CC2652R7 examples, but I'm hitting a compile error:
```
ninja: Entering directory `out/debug'
ninja: error: 'gen/drivers_cc13x4.a', needed by 'lib/libCHIP.a', missing and no known rule to make it
```
So it seems like support for CC2652R7 is slowly getting removed. Do you by any chance have access to CC2674R10?

Ryekia88, 2024-03-15:
I hope they aren't removing R7, we just got it. It still on sale for on their website. R10 looks to have more flash and ram. But all the pins are the same. Sorry it took so long to reply. I do 3-4 jobs here at a given time. I tried to get the nxp k32w1 to compile as well, being that there was a sample for the contact sensor. But both Ti and Nxp don't seem to work.

ReneJosefsen, 2024-03-18:
No worries. I have not had the time to look into the CC2652R7 build failure since we are internally in the process of transitioning to the CC2674R10. It is properly a good idea to reach out to TI to get an official statement on the state of CC2652R7 and Matter support, but my impression is that unfortunately it is being dropped due to the resource limitations.

I have no experience with NXP, so I can not comment on their setup, sorry.

Ryekia88, 2024-03-18:
Wouldn't the R7 and the R10 have same the same code? I wish there was a tutorial or something that could help me out. It's like the moment I catch on to something, another part confuses me. HAHA. This is not the language that I know. I have modified previous c++ code, but never really created my own. Well not to this level. I am trying to understand how the zap works.

Ryekia88, 2024-03-19:
I am using a google nest for my TBR. I doubt my job would allow me to buy another dev kit.

Ryekia88, 2024-03-19:
and now I have a new error. It says platform should be defined. But I didnt change anything.

ReneJosefsen, 2024-03-19:
CC2652 and CC2674 ser on different cores so the code is not the same. The pin out is so it is a "drop in replacement" on the hardware (as far as I understand)

I have not checked the latest SDK commit, but as mentioned I have the feeling that it is not working out of the box. Do you have any contact at TI you can reach out to to get the state of CC2652 confirmed by them?

Ryekia88, 2024-03-20:
No, anytime i email them they tell me to check the forums. haha