project-chip/connectedhomeip #31076: How can I use the ESP32 to open the bulb pairing window?

whqxbs, 2023-12-18:
### Build issue(s)

I've implemented using ESP32 to act as a switch to control the lights on/off with the help of clusters like binding, following the steps in `examples/light-switch-app/esp32` in Github. But I'm now trying to get to the button on the ESP32 to enable a pairing window that opens the bulb(through the cluster associated with the pairing function), just like a switch that I press and the bulb becomes available for pairing instead of on/off.
Is my idea realizable? Thanks！

bzbarsky-apple, 2023-12-20:
You use the OpenCommissioningWindow command.

closed 2023-12-20 by bzbarsky-apple

whqxbs, 2023-12-21:
No no,I think you misunderstood me. What I'm trying to accomplish is to use one of the ESP32 buttons to open the commissioning window of the light it's `BINDING` with, not the CHIP-TOOL command. That is, I want to realize that the device controls the device, not the console to control the device, do you understand what I mean?

For example, using `examples/light-switch-app/esp32` , when I press the on/off button, the light goes on or off.
What I want, when I press the on/off button, the light opens the commissioning window.

Looking forward to your reply.

bzbarsky-apple, 2023-12-22:
Ah, OK. Well, then you need:

1. The lights need to have their ACLs set so that the ESP32 is an administrator.
2. The ESP32 needs to send the OpenCommisioningWindow command to the lights when the button is pressed.