project-chip/connectedhomeip #31095: [1.3] The implementation of the Electrical Power Measurement Cluster is required

Ashwinigrl, 2023-12-19:
### Reproduction steps

Executed the below mentioned command to put DUT into a commissionable state
**./chip-all-clusters-app**

Once DUT reach the commissionable state sent below mentioned command on TH.
**./chip-tool pairing onnetwork 1 20202021**

After the successful commissioning, I attempted to validate the Electrical Power Measurement, but it is not implemented in chip-tool.

### Bug prevalence

Whenever I do

### GitHub hash of the SDK that was being used

7cbac9130d2b10d6bfdba873129dfbffa294cf8d

### Platform

raspi

### Platform Version(s)

1.3

### Type

Test Improvement

### Anything else?

Chip-tool has to be enabled with below mentioned Attribute:
PowerMode
Accuracy
Ranges
Voltage
Current

bzbarsky-apple, 2023-12-21:
This needs XML added to the SDK.

jamesharrow, 2024-01-24:
The XML and Cluster was implemented in PR #30389 and is in all-clusters-app
There is a new PR #31622 which enables it with fake test data in the energy-management-app

PR #31518 (2024-02-01): Feature/electrical energy & power measurement clusters
This PR adds the XML for the Electrical Energy and Power Measurement Clusters and basic implementations of servers.

This is based on https://github.com/CHIP-Specifications/connectedhomeip-spec/pull/7830

### Changes

- Added `src/app/zap-templates/zcl/data-model/chip/electrical-energy-measurement-cluster.xml`
- Added `src/app/zap-templates/zcl/data-model/chip/electrical-power-measurement-cluster.xml`
- Zap regen and updateall
- Added server side implementations in `src/app/clusters/electrical-energy-measurement-server/` and `src/app/clust [... cut]
PR #31616 (2024-02-12): Add Electrical power measurement to energy management app
This PR adds the EPM (Electrical Power Measurement) cluster into examples/energy-management-app.
This PR is based off #31518 and should not be reviewed until that has merged.
It is also based off #31622 which adds test event triggers to fake energy readings.

- [x] Add Delegate into EnergyManagementApp
- [x] Add helper code to demonstrate how to use the cluster
- [x] Add TE trigger to send power readings

Misc changes:
- [x] Added protection to ensure SafeAttributePersistenceProvider in EVSE cluster was initialised (relates to #31591 se [... cut]
closed 2024-02-01 by bzbarsky-apple