project-chip/connectedhomeip #31096: [1.3] The implementation of the Device Energy Management Cluster is required

Rajashreekalmane, 2023-12-19:
### Reproduction steps

Executed the below mentioned command to put DUT into a commissionable state
./chip-all-clusters-app

Once DUT reach the commissionable state sent below mentioned command on TH.
./chip-tool pairing onnetwork 1 20202021

After the successful commissioning, I attempted to validate the Microwave oven control Cluster , but it is not implemented in chip-tool

 ./chip-tool deviceenergymanagement read cluster-revision 1 1

> [1702963615.351626][6400:6402] CHIP:EM: Rxd Ack; Removing MessageCounter:217823681 from Retrans Table on exchange 12502i
> [1702963615.351727][6400:6402] CHIP:DMG: ReportDataMessage =
> [1702963615.351775][6400:6402] CHIP:DMG: {
> [1702963615.351813][6400:6402] CHIP:DMG: 	AttributeReportIBs =
> [1702963615.351872][6400:6402] CHIP:DMG: 	[
> [1702963615.351916][6400:6402] CHIP:DMG: 		AttributeReportIB =
> [1702963615.351977][6400:6402] CHIP:DMG: 		{
> [1702963615.352023][6400:6402] CHIP:DMG: 			AttributeStatusIB =
> [1702963615.352078][6400:6402] CHIP:DMG: 			{
> [1702963615.352129][6400:6402] CHIP:DMG: 				AttributePathIB =
> [1702963615.352186][6400:6402] CHIP:DMG: 				{
> [1702963615.352243][6400:6402] CHIP:DMG: 					Endpoint = 0x1,
> [1702963615.352305][6400:6402] CHIP:DMG: 					Cluster = 0x98,
> [1702963615.352367][6400:6402] CHIP:DMG: 					Attribute = 0x0000_FFFD,
> [1702963615.352424][6400:6402] CHIP:DMG: 				}
> [1702963615.352487][6400:6402] CHIP:DMG: 					
> [1702963615.352540][6400:6402] CHIP:DMG: 				StatusIB =
> [1702963615.352596][6400:6402] CHIP:DMG: 				{
> [1702963615.352653][6400:6402] CHIP:DMG: 					status = 0xc3 (UNSUPPORTED_CLUSTER),
> [1702963615.352711][6400:6402] CHIP:DMG: 				},
> [1702963615.352769][6400:6402] CHIP:DMG: 				
> [1702963615.352819][6400:6402] CHIP:DMG: 			},
> [1702963615.352876][6400:6402] CHIP:DMG: 			
> [1702963615.352921][6400:6402] CHIP:DMG: 		},
> [1702963615.352974][6400:6402] CHIP:DMG: 		
> [1702963615.353015][6400:6402] CHIP:DMG: 	],
> [1702963615.353068][6400:6402] CHIP:DMG: 	
> [1702963615.353111][6400:6402] CHIP:DMG: 	SuppressResponse = true, 
> [1702963615.353156][6400:6402] CHIP:DMG: 	InteractionModelRevision = 11
> [1702963615.353197][6400:6402] CHIP:DMG: }
> [1702963615.353467][6400:6402] CHIP:TOO: Response Failure: IM Error 0x000005C3: General error: 0xc3 (UNSUPPORTED_CLUSTER)
> [1702963615.353832][6400:6402] CHIP:EM: <<< [E:12502i S:53276 M:217823682 (Ack:84559124)] (S) Msg TX to 1:0000000000000001 [FFC7] [UDP:[fe80::e65f:1ff:fe49:ae1a%wlan0]:5540] --- Type 0000:10 (SecureChannel:StandaloneAck)

### Bug prevalence

whenever I do

### GitHub hash of the SDK that was being used

7cbac9130d2b10d6bfdba873129dfbffa294cf8d

### Platform

raspi

### Type

Test Improvement

bzbarsky-apple, 2023-12-21:
@Survensa Device Energy Management and DRLC are not the same thing.  Which one is this issue about?

Rajashreekalmane, 2023-12-22:
@bzbarsky-apple The issue is related to Device Energy management cluster.

bzbarsky-apple, 2024-01-02:
In that case, https://github.com/project-chip/connectedhomeip/pull/30957 will fix.