project-chip/connectedhomeip #31099: [1.3] The implementation of the Messages Cluster is required

Ashwinigrl, 2023-12-19:
### Reproduction steps

Executed the below mentioned command to put DUT into a commissionable state
**./chip-all-clusters-app**

Once DUT reach the commissionable state sent below mentioned command on TH.
./chip-tool pairing onnetwork 1 20202021

After the successful commissioning, I attempted to validate the  Messages Cluster, but it is not implemented in chip-tool.

### Bug prevalence

Whenever I do

### GitHub hash of the SDK that was being used

7cbac9130d2b10d6bfdba873129dfbffa294cf8d

### Platform

raspi

### Platform Version(s)

1.3

### Type

Test Improvement

### Anything else?

**Chip-tool has to be implement for below mentioned Attribute:**

- [ ] Messages
- [ ] ActiveMessageIDs

**Chip-tool has to be implement for below mentioned Commands:**

- [ ] PresentMessagesRequest
- [ ] CancelMessagesRequest

bzbarsky-apple, 2023-12-21:
This has nothing to do with chip-tool per se.  The XML for this cluster has not been added to the SDK.

Ashwinigrl, 2024-01-29:
Issue Fixed

closed 2024-01-29 by Ashwinigrl