project-chip/connectedhomeip #31101: [1.3] The implementation of the Thread Border Router Diagnostics is required .

Rajashreekalmane, 2023-12-19:
### Reproduction steps

Executed the below mentioned command to put DUT into a commissionable state
./chip-all-clusters-app

Once DUT reach the commissionable state sent below mentioned command on TH.
./chip-tool pairing onnetwork 1 20202021

After the successful commissioning, I attempted to validate the Thread Border Router Diagnostics cluster , but it is not implemented in chip-tool
[image]

### Bug prevalence

whenever I do 

### GitHub hash of the SDK that was being used

7cbac9130d2b10d6bfdba873129dfbffa294cf8d

### Platform

raspi

### Type

Test Improvement

bzbarsky-apple, 2023-12-21:
There is no XML for this cluster in the SDK.

bzbarsky-apple, 2023-12-21:
But also, this is not going to be part of 1.3, since it's totally unimplemented.