project-chip/connectedhomeip #31102: [1.2] cannot finish bootstrap.sh

mmt-10, 2023-12-19:
### Reproduction steps

1. git clone --recursive -b v1.2.0.1 https://github.com/project-chip/connectedhomeip.git matter-v1.2.0.1
2. cd matter-v1.2.0.1
3. source scripts/bootstrap.sh
  then, bootstrap.sh stopped and showed error.

### Bug prevalence

always

### GitHub hash of the SDK that was being used

commit 181b0cb14ff007ec912f2ba6627e05dfb066c008 (HEAD, tag: v1.2.0.1)

### Platform

core

### Type

Core SDK Crash

### Anything else?

log is here.
[v1.2.0.1-bootstrap.txt](https://github.com/project-chip/connectedhomeip/files/13712320/v1.2.0.1-bootstrap.txt)

mmt-10, 2023-12-19:
pip install --upgrade prompt-toolkit==3.0.43

closed 2023-12-19 by mmt-10

s-marios, 2023-12-20:
FWIW this also hit me, and doing

`pip install --upgrade prompt-toolkit==3.0.43`

as explained above fixes it for me too.