project-chip/connectedhomeip #31103: [1.3] chip-tool: There is no information about node subscription message

nmtoan91, 2023-12-19:
### Reproduction steps

The log message of the chip-tool for subscription look like this:
```
[1702975134.115338][194233:194235] CHIP:TOO: Endpoint: 1 Cluster: 0x0000_0006 Attribute 0x0000_0000 DataVersion: 1187244470
[1702975134.115353][194233:194235] CHIP:TOO:   OnOff: FALSE
```

However, the chip-tool paired to multiple on-off devices with different nodeId. How can I know the node that the notification comes from?

### Bug prevalence

all

### GitHub hash of the SDK that was being used

latest

### Platform

darwin

### Platform Version(s)

latest

### Type

Common Cluster Logic

bzbarsky-apple, 2023-12-21:
> How can I know the node that the notification comes from?

You can look at the actual RX log for the packet, which lists the node id...

nmtoan91, 2023-12-25:
@bzbarsky-apple 
I'm sorry but I couldn't find such info. Can you tell me more about this?

Here are the full logs on a notification message:
```
[1703488375.509688][240834:240835] CHIP:DL: Got IP address on interface: wlp3s0 IP: 192.168.11.14
[1703488375.509805][240834:240836] CHIP:DL: HandlePlatformSpecificBLEEvent 32770
[1703488376.124413][240834:240836] CHIP:EM: >>> [E:8144r S:46984 M:224700504] (S) Msg RX from 1:0000000000000D05 [83A4] --- Type 0001:05 (IM:ReportData)
[1703488376.124447][240834:240836] CHIP:EM: Handling via exchange: 8144r, Delegate: 0x559cc244c628
[1703488376.124493][240834:240836] CHIP:DMG: ReportDataMessage =
[1703488376.124506][240834:240836] CHIP:DMG: {
[1703488376.124520][240834:240836] CHIP:DMG:    SubscriptionId = 0x50203755,
[1703488376.124532][240834:240836] CHIP:DMG:    AttributeReportIBs =
[1703488376.124548][240834:240836] CHIP:DMG:    [
[1703488376.124559][240834:240836] CHIP:DMG:            AttributeReportIB =
[1703488376.124576][240834:240836] CHIP:DMG:            {
[1703488376.124588][240834:240836] CHIP:DMG:                    AttributeDataIB =
[1703488376.124605][240834:240836] CHIP:DMG:                    {
[1703488376.124619][240834:240836] CHIP:DMG:                            DataVersion = 0x46c3ea06,
[1703488376.124634][240834:240836] CHIP:DMG:                            AttributePathIB =
[1703488376.124649][240834:240836] CHIP:DMG:                            {
[1703488376.124664][240834:240836] CHIP:DMG:                                    Endpoint = 0x1,
[1703488376.124680][240834:240836] CHIP:DMG:                                    Cluster = 0x6,
[1703488376.124696][240834:240836] CHIP:DMG:                                    Attribute = 0x0000_0000,
[1703488376.124711][240834:240836] CHIP:DMG:                            }
[1703488376.124727][240834:240836] CHIP:DMG:
[1703488376.124744][240834:240836] CHIP:DMG:                            Data = false,
[1703488376.124758][240834:240836] CHIP:DMG:                    },
[1703488376.124774][240834:240836] CHIP:DMG:
[1703488376.124786][240834:240836] CHIP:DMG:            },
[1703488376.124803][240834:240836] CHIP:DMG:
[1703488376.124815][240834:240836] CHIP:DMG:    ],
[1703488376.124830][240834:240836] CHIP:DMG:
[1703488376.124843][240834:240836] CHIP:DMG:    InteractionModelRevision = 1
[1703488376.124854][240834:240836] CHIP:DMG: }
[1703488376.124929][240834:240836] CHIP:TOO: Endpoint: 1 Cluster: 0x0000_0006 Attribute 0x0000_0000 DataVersion: 1187244550
[1703488376.124946][240834:240836] CHIP:TOO:   OnOff: FALSE
[1703488376.124976][240834:240836] CHIP:DMG: Refresh LivenessCheckTime for 14224 milliseconds with SubscriptionId = 0x50203755 Peer = 01:0000000000000D05
[1703488376.125071][240834:240836] CHIP:EM: <<< [E:8144r S:46984 M:89627807 (Ack:224700504)] (S) Msg TX to 1:0000000000000D05 [83A4] [UDP:[fe80::5ee9:31ff:fef4:595%wlp3s0]:5540] --- Type 0001:01 (IM:StatusResponse)
[1703488376.129716][240834:240836] CHIP:EM: >>> [E:8144r S:46984 M:224700505 (Ack:89627807)] (S) Msg RX from 1:0000000000000D05 [83A4] --- Type 0000:10 (SecureChannel:StandaloneAck)
[1703488376.129741][240834:240836] CHIP:EM: Found matching exchange: 8144r, Delegate: (nil)
[1703488376.129757][240834:240836] CHIP:EM: Rxd Ack; Removing MessageCounter:89627807 from Retrans Table on exchange 8144r
```

bzbarsky-apple, 2024-01-02:
This part in your paste there:
```
[1703488376.124413][240834:240836] CHIP:EM: >>> [E:8144r S:46984 M:224700504] (S) Msg RX from 1:0000000000000D05 [83A4] --- Type 0001:05 (IM:ReportData)
```
says the DeportData was received from node `0xD05`, no?

nmtoan91, 2024-01-29:
@bzbarsky-apple thank you,
0xD05 is the node ID. However, Why the node id is not included in the callback
```
void BufferedReadCallback::OnAttributeData(const ConcreteDataAttributePath & aPath, TLV::TLVReader * apData,
                                           const StatusIB & aStatus)
```

bzbarsky-apple, 2024-02-02:
Basically because the assumption is that if you are doing a read, you know where you are doing it from....  Maybe it should have been included.  Hard to change that now.