project-chip/connectedhomeip #31106: [1.3]

Ashwinigrl, 2023-12-19:
### Reproduction steps

Executed the below mentioned command to put DUT into a commissionable state
**./chip-microwave-oven-app**

Once DUT reach the commissionable state sent below mentioned command on TH.
**./chip-tool pairing onnetwork 1 20202021**

After the successful commissioning Send the below command

`./chip-tool microwaveovencontrol read cook-time 1 1`

```
[1702986435.548379][14363:14365] CHIP:DMG: ReportDataMessage =
[1702986435.548384][14363:14365] CHIP:DMG: {
[1702986435.548387][14363:14365] CHIP:DMG: 	AttributeReportIBs =
[1702986435.548395][14363:14365] CHIP:DMG: 	[
[1702986435.548400][14363:14365] CHIP:DMG: 		AttributeReportIB =
[1702986435.548406][14363:14365] CHIP:DMG: 		{
[1702986435.548410][14363:14365] CHIP:DMG: 			AttributeDataIB =
[1702986435.548414][14363:14365] CHIP:DMG: 			{
[1702986435.548420][14363:14365] CHIP:DMG: 				DataVersion = 0xa8117ba3,
[1702986435.548423][14363:14365] CHIP:DMG: 				AttributePathIB =
[1702986435.548428][14363:14365] CHIP:DMG: 				{
[1702986435.548433][14363:14365] CHIP:DMG: 					Endpoint = 0x1,
[1702986435.548437][14363:14365] CHIP:DMG: 					Cluster = 0x5f,
**[1702986435.548442][14363:14365] CHIP:DMG: 					Attribute = 0x0000_0001,**
[1702986435.548445][14363:14365] CHIP:DMG: 				}
[1702986435.548452][14363:14365] CHIP:DMG: 					
[1702986435.548456][14363:14365] CHIP:DMG: 				Data = 30, 
[1702986435.548460][14363:14365] CHIP:DMG: 			},
[1702986435.548466][14363:14365] CHIP:DMG: 			
[1702986435.548469][14363:14365] CHIP:DMG: 		},
[1702986435.548476][14363:14365] CHIP:DMG: 		
[1702986435.548479][14363:14365] CHIP:DMG: 	],
[1702986435.548485][14363:14365] CHIP:DMG: 	
[1702986435.548489][14363:14365] CHIP:DMG: 	SuppressResponse = true, 
[1702986435.548493][14363:14365] CHIP:DMG: 	InteractionModelRevision = 11
[1702986435.548497][14363:14365] CHIP:DMG: }
```

As per Spec the CookTime attribute id is **0x0000**, but in sample app (Microwave-Oven-app) attribute Id is mentioned is **0x0001** 

### Bug prevalence

Whenever I do

### GitHub hash of the SDK that was being used

7cbac9130d2b10d6bfdba873129dfbffa294cf8d

### Platform

raspi

### Platform Version(s)

1.3

### Type

Test Improvement

### Anything else?

**Spec References:** 
- https://github.com/CHIP-Specifications/connectedhomeip-spec/blob/master/src/app_clusters/MicrowaveOvenControl.adoc#5-attributes

liangpy4, 2024-01-18:
Updated the attribute ID of cook-time in PR#31130.

Ashwinigrl, 2024-01-19:
Issue Fixed

closed 2024-01-19 by Ashwinigrl