project-chip/connectedhomeip #31111: [ICD] Figure out dynamic ICD detection.

yunhanw-google, 2023-12-19:
based upon chat with @bzbarsky-apple  We cannot browse text record..
Follow-up proposal:
1. Introduce the new attribute for LIT/SIT in ICDManagementCluster so that we can get the attribute change to know dynamic ICD via report.
2. To add some sort of subtype thing, similar to how we have _CM, because that can be browsed for, for example,  "BrowseNodes(DiscoveryType::kICD, filter);" 
not yet figure out the scenario "if we need to detect switches in both directions and distinguish them from the device just going off the network"

yunhanw-google, 2023-12-19:
@bhaskar-apple  what do you mean switch in both directions? It is SIT->LIT and LIT->SIT, yes, we need to detect both direction.
If ICD device is going off the network, we cannot find the SIT/LIT change from DNSSD, why do we need distinguish them?

yunhanw-google, 2023-12-19:
https://github.com/jmartinez-silabs/connectedhomeip/blob/e2421597b23272f10e481392acb99ae38889d477/src/lib/dnssd/Discovery_ImplPlatform.cpp#L240
@mkardous-silabs the change here for text record cannot be dynamic ICD detection..

yunhanw-google, 2023-12-20:
@bhaskar-apple  We need distinguish SIT->LIT from ICD device offline(for exmaple, power off), report in subscription in proposal 1 would guarantee that controller detect the change from SIT->LIT from proposed new attribute(type), if device is dead, controller cannot receive the report. In contrast, if applying proposal 2, when controller browsers the network, and ICD device already goes offline, controller cannot distinguish whether it is offline or SIT->LIT.

So it seems it is better to go with new type attribute in ICDManagement Cluster?

mkardous-silabs, 2024-01-04:
@yunhanw-google I think the best solution would be having both the DNS record and the attrbute value.
We the combination, we cover all the possible entry points.

1- Controller resolves ICD, it knows the operating mode
2- If the ICD changes operating mode, subs are updated and DNS record is update

yunhanw-google, 2024-01-04:
@mkardous-silabs I guess what you mean here is to combine DNS subtype and new attribute in ICD Management cluster, right

mkardous-silabs, 2024-01-05:
@yunhanw-google exactly

turon, 2024-01-08:
WG: subscription events should smooth over the need to detect dynamic changes to ICD state (SIT -> LIT or LIT -> SIT).

yunhanw-google, 2024-01-08:
@doru91 please prioritize this effort when you have time, thanks