project-chip/connectedhomeip #31115: Update system/SystemPacketBuffer to use uint32_t(instead of uint16_t) for several fields to accommodate large payloads

pidarped, 2023-12-19:
### Feature description

Currently, SystemPacketBuffer uses uint16_t for a lot of its internal and public parameters. These would need to be updated to accommodate large payload sizes when operating over TCP.
Internally, fields such as alloc_size uses uint16_t.
APIs such as MaxDataLength() and AvailableDataLength() return uint16_t. These need to use uint32_t when handling large sized buffers that go beyond uint16_t limits.

### Platform

core

bzbarsky-apple, 2023-12-20:
Shouldn't it just use size_t for the APIs?

pidarped, 2024-01-29:
@bzbarsky-apple, I had considered using size_t but on second thoughts felt that uint32_t might be better. Since the max size of the field in the TCP payload is set to 4 bytes, everything in the code, from processing to storage, should be dictated by that, IMO.
Changes have to be made to both the API arguments as well as internal storage. Using uint32_t at all the relevant places seems to require less static_casts. Also, size_t, being not the same in all architectures(8 bytes in Linux standalone) might pose weird problems(during casting in and out of it) in some edge cases, that I would want to avoid.

pidarped, 2024-05-05:
Fixed by #33001 in master.

PR #31776 (2024-02-23): Adjust storage and processing in SystemPacketBuffer and associated code for large payloads.
Extend uint16_t type variables to uint32_t wherever applicable.

Fixes #31115
closed 2024-05-05 by pidarped