project-chip/connectedhomeip #31116: About Profile ID 0x0103 in ZAP tool

chenek, 2023-12-19:
### Reproduction steps

Start ZAP tool will see this

### Bug prevalence

Every time

### GitHub hash of the SDK that was being used

Latest

### Platform

efr32, windows

### Anything else?

As we can see Profile ID 0x0103 in ZAP tool, I suppose Profile ID 0x0103 is specific for Matter applications but I cannot find its definition anywhere in Matter specifications. Can anyone confirm this and point to where this profile ID is defined?

bzbarsky-apple, 2023-12-20:
Profile IDs are ZAP-specific.  There is no concept of "profile id" in the way ZAP means it in Matter.  Please file issues on ZAP at https://github.com/project-chip/zap/issues

chenek, 2023-12-20:
Thanks, I’ll post this issue at [https://github.com/project-chip/zap/issues](url)