project-chip/connectedhomeip #31122: [1.3] Microwave Oven Control Cluster Feature-map value mismatch.

Ashwinigrl, 2023-12-20:
### Reproduction steps

Executed the below mentioned command to put DUT into a commissionable state
**./chip-microwave-oven-app**

Once DUT reach the commissionable state sent below mentioned command on TH.
**./chip-tool pairing onnetwork 1 20202021**

After the successful commissioning Send the below command

**Actual behavior:**

./chip-tool microwaveovencontrol read feature-map 1 1

```
[1702972998.240172][9703:9705] CHIP:DMG: }
[1702972998.240301][9703:9705] CHIP:TOO: Endpoint: 1 Cluster: 0x0000_005F Attribute 0x0000_FFFC DataVersion: 486770894
[1702972998.240343][9703:9705] CHIP:TOO:   FeatureMap: 0
```

**Expected behavior:**

./chip-tool microwaveovencontrol read feature-map 1 1

```
[1702972998.240172][9703:9705] CHIP:DMG: }
[1702972998.240301][9703:9705] CHIP:TOO: Endpoint: 1 Cluster: 0x0000_005F Attribute 0x0000_FFFC DataVersion: 486770894
[1702972998.240343][9703:9705] CHIP:TOO:   FeatureMap: 1
```

### Bug prevalence

Whenever I do

### GitHub hash of the SDK that was being used

7cbac9130d2b10d6bfdba873129dfbffa294cf8d

### Platform

raspi

### Platform Version(s)

1.3

### Type

Test Improvement

### Anything else?

In Microwave Oven Control Cluster Feature Map value should be 1, But Currently DUT is responding with value 0. `PWRNUM `feature is set to true, because the attribute `PowerSetting` is depending on the feature `PWRNUM` is implemented. So Feature Map value should be 1. 

**Specification References:**

- https://github.com/CHIP-Specifications/connectedhomeip-spec/blob/master/src/app_clusters/MicrowaveOvenControl.adoc#4-features

bzbarsky-apple, 2023-12-20:
@fengdx4 this seems like a bug in https://github.com/project-chip/connectedhomeip/pull/30235

liangpy4, 2024-01-18:
Feature-map in the MicrowaveOven example app is set to 5 now. You may close this issue.

Ashwinigrl, 2024-01-19:
Issue Fixed

closed 2024-01-19 by Ashwinigrl