project-chip/connectedhomeip #31125: [BUG] chip-tool problem with Persistent Storage on MacOs

RJ030, 2023-12-20:
### Reproduction steps

Using chip-tool, I perform a commissioning of my WiFi device (a temperature-measurement device):
`out/chiptoolex/chip-tool pairing ble-wifi 1234 WIFI_SSID WIFI_PWD 20202021 3840`
The commissioning is working, a Fabric is created.

After the commissioning, I can read the temperature measurement using chip-tool:
`out/chiptoolex/chip-tool temperaturemeasurement read measured-value 1234 1`

If I reboot the Mac, I'm not able to read the measurement anymore. I believe the problem comes from the FabricTable::Init method (src/credentials/FabricTable.cpp). Since chip-tool does not found a Fabric, it creates a new one, and then cannot resolve the id.
`CHIP_ERROR err = mStorage->SyncGetKeyValue(DefaultStorageKeyAllocator::FabricIndexInfo().KeyName(), buf, size);` returns `CHIP_ERROR_PERSISTED_STORAGE_VALUE_NOT_FOUND`

### Bug prevalence

Every time after a reboot

### GitHub hash of the SDK that was being used

1f3bcbc955522430151d3127b5dbc7140b7a2366

### Platform

other

### Platform Version(s)

macOS Sonoma Version 14.1.1

### Anything else?

Log:
out/chiptoolex/chip-tool temperaturemeasurement read measured-value 1234 1
[1702891582541] [2407:18271] [DL] KVS will be written to: [file:///var/folders/5k/7q2j5t6n6pv5kwflks_fb9600000gp/T//chip_tool_kvs](file:///var/folders/5k/7q2j5t6n6pv5kwflks_fb9600000gp/T/chip_tool_kvs)
[1702891582567] [2407:18271] [DL] _Init
[1702891582567] [2407:18271] [IN] UDP::Init bind&listen port=0
[1702891582567] [2407:18271] [IN] UDP::Init bound to port=60464
[1702891582567] [2407:18271] [IN] UDP::Init bind&listen port=0
[1702891582567] [2407:18271] [IN] UDP::Init bound to port=59896
[1702891582567] [2407:18271] [IN] BLEBase::Init - setting/overriding transport
[1702891582567] [2407:18271] [IN] TransportMgr initialized
[1702891582568] [2407:18271] [FP] Initializing FabricTable from persistent storage
[1702891582568] [2407:18271] [TS] Last Known Good Time: [unknown]
[1702891582568] [2407:18271] [TS] Setting Last Known Good Time to firmware build time 2023-10-14T01:16:48
[1702891582571] [2407:18271] [ZCL] Using ZAP configuration...
[1702891582571] [2407:18271] [CTL] System State Initialized...
[1702891582572] [2407:18271] [CTL] Setting attestation nonce to random value
[1702891582572] [2407:18271] [CTL] Setting CSR nonce to random value
[1702891582572] [2407:18271] [IN] UDP::Init bind&listen port=5550
[1702891582572] [2407:18271] [IN] UDP::Init bound to port=5550
[1702891582572] [2407:18271] [IN] UDP::Init bind&listen port=5550
[1702891582572] [2407:18271] [IN] UDP::Init bound to port=5550
[1702891582572] [2407:18271] [IN] TransportMgr initialized
[1702891582572] [2407:18274] [DL] _OnPlatformEvent
[1702891582573] [2407:18274] [CTL] Setting attestation nonce to random value
[1702891582573] [2407:18274] [CTL] Setting CSR nonce to random value
[1702891582573] [2407:18274] [CTL] Couldn't get ExampleOpCredsCAKey from storage: src/controller/ExamplePersistentStorage.cpp:106: CHIP Error 0x000000A0: Value not found in the persisted storage
[1702891582575] [2407:18274] [CTL] Couldn't get ExampleOpCredsICAKey from storage: src/controller/ExamplePersistentStorage.cpp:106: CHIP Error 0x000000A0: Value not found in the persisted storage
[1702891582578] [2407:18274] [CTL] Generating RCAC
[1702891582581] [2407:18274] [CTL] Generating ICAC
[1702891582582] [2407:18274] [CTL] Generating NOC
[1702891582583] [2407:18274] [FP] Validating NOC chain
[1702891582587] [2407:18274] [FP] NOC chain validation successful
[1702891582587] [2407:18274] [FP] Added new fabric at index: 0x1
[1702891582587] [2407:18274] [FP] Assigned compressed fabric ID: 0x3AC46FE94CE9DABF, node ID: 0x000000000001B669
[1702891582587] [2407:18274] [TS] Last Known Good Time: 2023-10-14T01:16:48
[1702891582587] [2407:18274] [TS] New proposed Last Known Good Time: 2021-01-01T00:00:00
[1702891582587] [2407:18274] [TS] Retaining current Last Known Good Time
[1702891582589] [2407:18274] [FP] Metadata for Fabric 0x1 persisted to storage.
[1702891582590] [2407:18274] [TS] Committing Last Known Good Time to storage: 2023-10-14T01:16:48
[1702891582592] [2407:18274] [CTL] Joined the fabric at index 1. Fabric ID is 0x0000000000000001 (Compressed Fabric ID: 3AC46FE94CE9DABF)
[1702891582592] [2407:18274] [IN] UDP::Init bind&listen port=5550
[1702891582592] [2407:18274] [IN] UDP::Init bound to port=5550
[1702891582592] [2407:18274] [IN] UDP::Init bind&listen port=5550
[1702891582592] [2407:18274] [IN] UDP::Init bound to port=5550
[1702891582592] [2407:18274] [IN] TransportMgr initialized
[1702891582608] [2407:18274] [TOO] Sending command to node 0x4d2
[1702891582608] [2407:18274] [CSM] FindOrEstablishSession: PeerId = [1:00000000000004D2]
[1702891582608] [2407:18274] [CSM] FindOrEstablishSession: No existing OperationalSessionSetup instance found
[1702891582608] [2407:18274] [DIS] OperationalSessionSetup[1:00000000000004D2]: State change 1 --> 2
[1702891582608] [2407:18274] [DIS] Resolving 3AC46FE94CE9DABF:00000000000004D2 ...
[1702891582609] [2407:18274] [DIS] Resolve type=_matter._tcp name=3AC46FE94CE9DABF-00000000000004D2 interface=0
[1702891582809] [2407:18274] [DIS] Checking node lookup status after 200 ms
[1702891627608] [2407:18275] [DIS] Checking node lookup status after 45002 ms
[1702891627608] [2407:18275] [DIS] No longer need resolve for 3AC46FE94CE9DABF-00000000000004D2
[1702891627609] [2407:18275] [DIS] Mdns: Resolve failure (src/platform/Darwin/DnssdImpl.cpp:476: CHIP Error 0x00000074: The operation has been cancelled)
[1702891627609] [2407:18275] [DIS] OperationalSessionSetup[1:00000000000004D2]: operational discovery failed: src/lib/address_resolve/AddressResolve_DefaultImpl.cpp:119: CHIP Error 0x00000032: Timeout
[1702891627609] [2407:18275] [-] src/lib/address_resolve/AddressResolve_DefaultImpl.cpp:119: CHIP Error 0x00000032: Timeout at ../../examples/chip-tool/commands/clusters/ModelCommand.cpp:62
[1702891627609] [2407:18271] [CTL] Shutting down the commissioner
[1702891627609] [2407:18271] [CTL] Shutting down the controller
[1702891627609] [2407:18271] [IN] Expiring all sessions for fabric 0x1!!
[1702891627609] [2407:18271] [FP] Forgetting fabric 0x1
[1702891627609] [2407:18271] [TS] Pending Last Known Good Time: 2023-10-14T01:16:48
[1702891627609] [2407:18271] [TS] Previous Last Known Good Time: 2023-10-14T01:16:48
[1702891627609] [2407:18271] [TS] Reverted Last Known Good Time to previous value
[1702891627609] [2407:18271] [CTL] Shutting down the commissioner
[1702891627609] [2407:18271] [CTL] Shutting down the controller
[1702891627609] [2407:18271] [CTL] Shutting down the System State, this will teardown the CHIP Stack
[1702891627609] [2407:18271] [DMG] All ReadHandler-s are clean, clear GlobalDirtySet
[1702891627609] [2407:18271] [BLE] ConnectionDelegate CancelConnection
[1702891627610] [2407:18271] [FP] Shutting down FabricTable
[1702891627610] [2407:18271] [TS] Pending Last Known Good Time: 2023-10-14T01:16:48
[1702891627610] [2407:18271] [TS] Previous Last Known Good Time: 2023-10-14T01:16:48
[1702891627610] [2407:18271] [TS] Reverted Last Known Good Time to previous value
[1702891627610] [2407:18271] [DL] Inet Layer shutdown
[1702891627610] [2407:18271] [DL] BLE shutdown
[1702891627610] [2407:18271] [DL] System Layer shutdown
[1702891627610] [2407:18271] [TOO] Run command failure: src/lib/address_resolve/AddressResolve_DefaultImpl.cpp:119: CHIP Error 0x00000032: Timeout

bzbarsky-apple, 2023-12-20:
By default chip-tool puts its persistent storage in `/tmp`.  Rebooting the machine then clears the storage.

You can specify a different `--storage-directory` on the chip-tool command line (or set TMPDIR) if you want to use some other place to store the data.

closed 2023-12-20 by bzbarsky-apple

RJ030, 2023-12-20:
Hello bzbarsky, thanks for your help. I thought that the /var directory was not flushed during rebooting but it is...

I will add this information inside the "working with the Chip-tool" readme file. Maybe it will help other users.