project-chip/connectedhomeip #31126: [BUG] [ESP32C3][lock-app] Chip stack locking error - code is unsafe/racey

s-marios, 2023-12-20:
### Reproduction steps

Same as #25874 

The log of the application crashing, when scheduling work to be done.

[esp32c3_crash.log](https://github.com/project-chip/connectedhomeip/files/13724480/esp32c3_crash.log)

### Bug prevalence

always

### GitHub hash of the SDK that was being used

181b0cb14ff007ec912f2ba6627e05dfb066c008 (tag: v1.2.0.1)

### Platform

esp32

### Platform Version(s)

v1.2.0.1, master

### Anything else?

A simple fix is to lock the chip stack before submitting new work to be done. The following patch fixes the problem as far as I can tell.

```
$: cat 0001-Lock-chip-core-before-submitting-tasks.patch 

From ff18fc7f90f426901de1e429e3143d3ac6820f27 Mon Sep 17 00:00:00 2001
From: "Marios S@pad" <smarios@jaist.ac.jp>
Date: Wed, 20 Dec 2023 16:02:47 +0900
Subject: [PATCH] Lock chip core before submitting tasks

---
 examples/lock-app/esp32/main/AppTask.cpp | 2 ++
 1 file changed, 2 insertions(+)

diff --git a/examples/lock-app/esp32/main/AppTask.cpp b/examples/lock-app/esp32/main/AppTask.cpp
index 8b251a3..0b34466 100644
--- a/examples/lock-app/esp32/main/AppTask.cpp
+++ b/examples/lock-app/esp32/main/AppTask.cpp
@@ -424,6 +424,8 @@ void AppTask::ActionCompleted(BoltLockManager::Action_t aAction)
     }
     if (sAppTask.mSyncClusterToButtonAction)
     {
+        chip::DeviceLayer::StackLock lock;
+
         chip::DeviceLayer::SystemLayer().ScheduleWork(UpdateClusterState, nullptr);
         sAppTask.mSyncClusterToButtonAction = false;
     }
-- 
2.43.0

```

Again, this is a problem which in essence is the same to #25874 , it just occurred at a different place.

bzbarsky-apple, 2023-12-20:
@dhrishi @shubhamdp

shubhamdp, 2023-12-22:
@s-marios I build the app on master and commissioned the device and sent `lock-door` and `unlock-door` command, but the device did not crash.

But there is unsafe access to chip-stack though and I have fixed it #31171. Please check if this patch fixes the crash for you.

closed 2023-12-22 by mergify[bot]

s-marios, 2023-12-25:
> @s-marios I build the app on master and commissioned the device and sent `lock-door` and `unlock-door` command, but the device did not crash.

I'm surprised that on your end the crash would not manifest. On my end/configuration it is a 100% occurrence.
> 
> But there is unsafe access to chip-stack though and I have fixed it #31171. Please check if this patch fixes the crash for you.

I haven't had the time to apply the fix yet, but it looks like it addresses the locking of the chip stack.

I'll report back as time permits!

PR #31171 (2023-12-22): [ESP32] fix the unsafe access to chip stack in lock-app
Fixes #31126

`SystemLayer::ScheduleWork()` must be called from chip thread and `PlatformMgr::ScheduleWork()` can be called from any thread.

Replaced the `SystemLayer::ScheduleWork()` with `PlatformMgr::ScheduleWork()` for thread safety.