project-chip/connectedhomeip #31129: [Platform] Failed encoding invalid payload with setupCommissioningSession

law-ko, 2023-12-20:
### Reproduction steps

We are trying to commission a device without using QRcode, but Matter returns invalid payload, which we do not know what information is missing.

Setup:
```
self.setupPayload = MTRSetupPayload(setupPasscode: passcode, discriminator: NSNumber(value: 3840))
self.matterController!.setDeviceControllerDelegate(self, queue: self.queue)
let deviceId = MTRGetNextAvailableDeviceID()
MTRSetNextAvailableDeviceID(deviceId+1)
if self.setupPayload != nil
    && self.matterController != nil {
    try self.matterController!.setupCommissioningSession(
        with: self.setupPayload!,
        newNodeID: NSNumber(value: deviceId))
}
```

The log shows:
```
self.setupPayload.setupPasscode:  Optional(531748882)
self.setupPayload.discriminator:  Optional(3840)
self.setupPayload.productID:  Optional(0)
self.setupPayload.vendorID:  Optional(0)
★deviceId: 9
2023-12-20 17:17:25.235487+0800 Invis[45799:2887819] [SetupPayload] Failed encoding invalid payload
2023-12-20 17:17:25.235588+0800 Invis[45799:2887819] [Default] Error(/Library/Caches/com.apple.xbs/Sources/CHIPFramework/connectedhomeip/src/darwin/Framework/CHIP/MTRDeviceController.mm:394: CHIP Error 0x0000002F: Invalid argument): Generating Manual Pairing Code failed
Error Domain=MTRErrorDomain Code=4 "An argument is invalid." UserInfo={NSLocalizedDescription=An argument is invalid., underlyingError=<MTRErrorHolder: 0x28372dfe0>}
```

Why does generating the manual pairing code failed?

Thank you.

### Platform

darwin

### Platform Version(s)

16.4

### Type

Manually tested with SDK

### (Optional) If manually tested please explain why this is only manually tested

Tested on iPhone with 16.4.1

law-ko, 2023-12-20:
@bzbarsky-apple We have two senarios:

1. If the device is never being commissioned, then the user may enter pairing code instead of scanning.
2. If the device is already being commissioned (which is currently connected to a Wi-Fi network), then the user may want to add the device onto our custom app, which uses the `Turn on Pairing Mode` or [`openCommissioningWindow`](https://developer.apple.com/documentation/matter/mtrbasedevice/4110507-opencommissioningwindow?changes=latest_minor) code that generates the passcode.

We are unsure which function to use and how to pass in the pairing code/passcode.

bzbarsky-apple, 2023-12-20:
@law-ko For that first scenario, when a user enters a "pairing code" (I assume you mean a Matter numeric code), you can create a MTRSetupPayload directly from it, right?  Something like:
```
self.setupPayload = MTRSetupPayload(onboardingPayload: whateverUserEntered)
```
That works for both QR codes and numeric codes.

For the second scenario... I am confused about what the question is.  If you use the API you linked to, then the MTRSetupPayload you should for setting up the commissioning session use will be returned from the API, no? 

> We are unsure which function to use and how to pass in the pairing code/passcode.

Just so we are on the same page: those are not the same thing, right?  A pairing code is either a QR code or numeric code that encodes the passcode and some other information (e.g. the discriminator).

bzbarsky-apple, 2023-12-20:
In general, you should probably not be using `MTRSetupPayload(setupPasscode: ....)` at all.

law-ko, 2023-12-21:
@bzbarsky-apple 
For the second scenario (like in attached image), what MTR function should we use?
[image]

law-ko, 2023-12-21:
> @bzbarsky-apple
> For the second scenario (like in attached image), what MTR function should we use?
> <img alt="image" width="270" src="https://private-user-images.githubusercontent.com/105698389/292062646-0dd9521b-1d07-4d31-ad71-2d486cbd56e3.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTEiLCJleHAiOjE3MDMxMjg3NDQsIm5iZiI6MTcwMzEyODQ0NCwicGF0aCI6Ii8xMDU2OTgzODkvMjkyMDYyNjQ2LTBkZDk1MjFiLTFkMDctNGQzMS1hZDcxLTJkNDg2Y2JkNTZlMy5wbmc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBSVdOSllBWDRDU1ZFSDUzQSUyRjIwMjMxMjIxJTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDIzMTIyMVQwMzE0MDRaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT01MjA3N2RjMGE5ZTY5NDUwMGQ1YjFlOGFmOTFlOTg5YTBkZTJmN2I1YjY1NDBhYWNjNTAyMmM0YzZmNzM3N2MxJlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCZhY3Rvcl9pZD0wJmtleV9pZD0wJnJlcG9faWQ9MCJ9.ilCU8N5v6ZGFosNXwk-8xbZ7tQj2V3EuB48PRQt35qw">

@bzbarsky-apple Same function `MTRSetupPayload(onboardingPayload: self.payloadCode)` would work. However, is there any way to know the device is already commissioned (connected to the Wi-Fi network) so the app can skip the entering Wi-Fi credentials?

closed 2023-12-22 by mergify[bot]

law-ko, 2023-12-22:
> @bzbarsky-apple Same function `MTRSetupPayload(onboardingPayload: self.payloadCode)` would work. However, is there any way to know the device is already commissioned (connected to the Wi-Fi network) [...]

@bzbarsky-apple I see this issue is closed, are you able to provide some information that I asked above?

bzbarsky-apple, 2023-12-22:
@law-ko If you know the device is on the network already, setting the discoveryCapabilities to on-network only should work, I would think...

law-ko, 2023-12-22:
@bzbarsky-apple Is there any chance to know if the device is on the network already based on any MTR commands or based on the payloadCode?

bzbarsky-apple, 2023-12-22:
If you have a QR code, then "maybe", since that can indicate how the device is discoverable.

With a numeric code, no.

I'm not sure what you mean by "MTR commands".  Do you mean "Matter framework APIs that do not involve having a secure connection to the device"?  If so, there are no such APIs that can tell you the device is on-network.