project-chip/connectedhomeip #31135: [BUG] ESP32-C6 lighting demo pairing ble-thread fails

Sjors, 2023-12-20:
### Reproduction steps

The device is a ESP32-C6-DevKitC-1 v1.2

I'm unable to get past the pairing stage. It fails with `BLE GAP connection terminated (con 1 reason 0x213)`.

Using ESP-IDF version v5.1.2 on Intel macOS 14.2, Python 3.12.1, first I tried the [hello world]( https://docs.espressif.com/projects/esp-idf/en/v5.1.2/esp32c6/get-started/linux-macos-setup.html) app, which works.

I run Home Assistant the OpenThread Border Router and Matter addon. It runs on channel 15. For the "operational-dataset" value I copied the "Active dataset TLVs" field that was hidden under Settings -> Thread -> Configure -> Prefered Network -> Info.

The `chip-tool pairing ble-thread 1 hex:0e0800000...f7f8 20202021 3840` command fails with `Ble Error 0x00000407: GATT write characteristic operation failed`.

Additional logs below, happy to add more. 

Following the steps of [esp32/setup_idf_chip](https://github.com/project-chip/connectedhomeip/blob/master/docs/guides/esp32/setup_idf_chip.md) as closely as possible.

In `examples/lighting-app/esp32` I run:

```
idf.py -p /dev/cu.usbmodem14201 flash monitor
Executing action: flash
Running cmake in directory ...
...
I (23) boot: ESP-IDF v5.1.2 2nd stage bootloader
I (23) boot: compile time Dec 20 2023 15:51:58
I (24) boot: chip revision: v0.0
I (26) qio_mode: Enabling default flash chip QIO
I (31) boot.esp32c6: SPI Speed      : 40MHz
I (36) boot.esp32c6: SPI Mode       : QIO
I (41) boot.esp32c6: SPI Flash Size : 4MB
I (45) boot: Enabling RNG early entropy source...
I (51) boot: Partition Table:
I (54) boot: ## Label            Usage          Type ST Offset   Length
I (62) boot:  0 esp_secure_cert  unknown          3f 06 0000d000 00002000
I (69) boot:  1 nvs              WiFi data        01 02 00010000 0000c000
I (76) boot:  2 nvs_keys         NVS keys         01 04 0001c000 00001000
I (84) boot:  3 otadata          OTA data         01 00 0001d000 00002000
I (91) boot:  4 phy_init         RF data          01 01 0001f000 00001000
I (99) boot:  5 ota_0            OTA app          00 10 00020000 001e0000
I (106) boot:  6 ota_1            OTA app          00 11 00200000 001e0000
I (114) boot:  7 fctry            WiFi data        01 02 003e0000 00006000
I (122) boot: End of partition table
I (126) boot: No factory image, trying OTA 0
I (131) esp_image: segment 0: paddr=00020020 vaddr=42130020 size=3b780h (243584) map
I (191) esp_image: segment 1: paddr=0005b7a8 vaddr=40800000 size=04870h ( 18544) load
I (196) esp_image: segment 2: paddr=00060020 vaddr=42000020 size=127ee0h (1212128) map
I (455) esp_image: segment 3: paddr=00187f08 vaddr=40804870 size=17bd4h ( 97236) load
I (485) boot: Loaded app from partition at offset 0x20000
I (512) boot: Set actual ota_seq=1 in otadata[0]
I (512) boot: Disabling RNG early entropy source...
I (523) cpu_start: Unicore app
I (523) cpu_start: Pro cpu up.
W (532) clk: esp_perip_clk_init() has not been implemented yet
I (538) cpu_start: Pro cpu start user code
I (539) cpu_start: cpu freq: 160000000 Hz
I (539) cpu_start: Application information:
I (541) cpu_start: Project name:     chip-lighting-app
I (547) cpu_start: App version:      v1.0
I (552) cpu_start: Compile time:     Dec 20 2023 17:36:19
I (558) cpu_start: ELF file SHA256:  3fab16b968f54c7e...
I (564) cpu_start: ESP-IDF:          v5.1.2
I (569) cpu_start: Min chip rev:     v0.0
I (574) cpu_start: Max chip rev:     v0.99 
I (578) cpu_start: Chip rev:         v0.0
I (583) heap_init: Initializing. RAM available for dynamic allocation:
I (590) heap_init: At 40830170 len 0004C4A0 (305 KiB): D/IRAM
I (597) heap_init: At 4087C610 len 00002F54 (11 KiB): STACK/DIRAM
I (604) heap_init: At 50000000 len 00003FE8 (15 KiB): RTCRAM
I (611) spi_flash: detected chip: generic
I (615) spi_flash: flash io: qio
[... 259 lines cut ...]
I (60711) NimBLE: GAP procedure initiated: advertise; 
I (60711) NimBLE: disc_mode=2
I (60711) NimBLE:  adv_channel_map=0 own_addr_type=1 adv_filter_policy=0 adv_itvl_min=40 adv_itvl_max=40
I (60721) NimBLE: 

I (60731) chip[DL]: CHIPoBLE advertising started
I (60731) app-devicecallbacks: CHIPoBLE disconnected
I (60731) app-devicecallbacks: Current free heap: 219280
```

### Bug prevalence

Always

### GitHub hash of the SDK that was being used

19e202e62f3051f62e720412f439c4777405485d

### Platform

esp32

### Platform Version(s)

5.1.2

Sjors, 2023-12-20:
By the way, my goal is to get Matter via Thread to work, but as an intermediate step Matter via Wifi is fine too.

bzbarsky-apple, 2023-12-21:
@Sjors Your issue is here:
``
[1703090316035] [62745:13051558] [BLE] BLE:Error writing Characteristics in Chip service on the device: [The specified UUID is not allowed for this operation.]
```

To use chip-tool with BLE on Mac, please follow the directions at https://github.com/project-chip/connectedhomeip/blob/master/docs/guides/darwin.md#using-chip-tool-on-macos-or-chip-tool-on-ios.  In particular, you have to have the Bluetooth Central Matter Client Developer mode profile installed.

closed 2023-12-21 by bzbarsky-apple

Sjors, 2023-12-21:
@bzbarsky-apple that's, I'll give that a try. But I'm confused: why would I need any proprietary (?) Apple certificate in order to connect an ESP32 with Home Assistant? Neither of these have anything to do with Apple. Only the laptop I use to the commissioning from happens to be a Macbook, but might as well be an Ubuntu machine.

bzbarsky-apple, 2023-12-22:
The Matter BLE UUID is blocked by default on Apple devices, in the Bluetooth subsystem.   The profile install un-blocks it.

I can't give you details on why that is, sorry.

Sjors, 2023-12-22:
@bzbarsky-apple thanks, that makes sense then. So if I understand correctly I wouldn't need to this when using a non-apple device.

bzbarsky-apple, 2024-01-03:
Correct, but note that "device" in this case means "laptop".