project-chip/connectedhomeip #31154: [BUG] Thermostat setpoint lower raise command

doublemis1, 2023-12-21:
### Reproduction steps

Default Max Cool/Heat Setpoint:
Heat = 3000
Cool = 3200

Commission Thermostat
Set occupancy heating setpoint to 2000: ./chip-tool thermostart write occupied-heating-setpoint 2000 1 1
Set occupancy cooling setpoint to 2600: ./chip-tool thermostart write occupied-cooling-setpoint 2600 1 1
Run command to increase of 1200 both values: ./chip-tool thermostat setpoint-raise-lower 2 120 1 1
Read occupancy cooling setpoint ./chip-tool thermostart read occupied-cooling-setpoint 1 1→ 3200
Read occupancy heating setpoint ./chip-tool thermostart read occupied-heating-setpoint 1 1→ expected 3000, got 2600(failure)

### Bug prevalence

always

### GitHub hash of the SDK that was being used

master

### Platform

other

### Anything else?

Based on the specification (4.3.10.1):
Upon receipt, the attributes for the indicated setpoint(s) SHALL have the amount specified in the Amount field added to them. If the resulting value is outside the limits imposed by MinCoolSet­pointLimit, MaxCoolSetpointLimit, MinHeatSetpointLimit and MaxHeatSetpointLimit, the value is clamped to those limits. This is not considered an error condition.

However if the the device will change the occupancy setpoint bot on cooling and heating the value should be clamped to the limits:
e.g.
Max Heat Setpoint Limit = 3000
Max Cool Setpoint Limit = 3200
Occupied Heating Setpoint = 2000
Occupied Cooling Setpoint = 2600
if run command ./chip-tool thermostat setpoint-raise-lower 2 120 1 1 where
2 - Setpoint Adjust Mode BOTH (Heat and Cool)
120 - is an amount which means increase 1200  

The final occupancy will be set to:
Occupancy Cooling Setpoint  = 3200 = Max Cool Setpoint Limit (600 increased)
Occupancy Heating Setpoint = 2600 !=  Max Heat Setpoint Limit (600 increased, should 1000)

doublemis1, 2024-01-18:
Reproduced on linux all-cluster-app

KishokG, 2024-01-23:
@doublemis1 As per understanding, The AUTO feature is implemented in the Thermostat-app . So the Max Heat Setpoint Limit may change according to the MinSetpointDeadBand.

Please correct me if I am wrong

doublemis1, 2024-04-03:
Hi, I was testing it with dead band and it also not work taking into account dead band